RobotLocomotion/drake #20713: AWS Backup Rules

svenevs, 2023-12-30:
drake-webdav is not currently getting backed up, discovered as part of looking at the jenkins setups.

This should get fixed... rule is created, will need to confirm viability of backup after it is created (daily in the [same group as the other backups](https://us-east-1.console.aws.amazon.com/backup/home?region=us-east-1#/backupplan/details/83a5c41b-0f95-47f3-bac0-af87484543be)).

CC @jwnimmer-tri @BetsyMcPhail

jwnimmer-tri, 2023-12-30:
To be clear -- is this regarding the server software itself (e.g., things like an nginx config file, ssh setup, etc.)?

In particular, note that we do _not_ need to back up the cached data itself (the files bazel writes into the cache).

svenevs, 2023-12-30:
> To be clear -- is this regarding the server software itself (e.g., things like an nginx config file, ssh setup, etc.)?

Yes, the instance itself.  Previously we had a backup routine hooked up to the _old_ (removed months ago at this point) `drake-webdav` instance and so there were a number of [failing backups](https://us-east-1.console.aws.amazon.com/backup/home?region=us-east-1#/jobs/backup?period=LAST_14_DAYS&status=FAILED) related to this.

TL;DR: from my perspective, these are very useful to keep because you can restore directly to a new instance and switch over the elastic IP to become operational within minutes, provided you remember to (now that it's on a new instance) fix the backup rules.

> note that we do not need to back up the cached data itself

Whoops ... yes you're absolutely right, I deleted that resource assignment and created a new one to just backup the instance.  Currently we have [a drake backup plan](https://us-east-1.console.aws.amazon.com/backup/home?region=us-east-1#/backupplan/details/83a5c41b-0f95-47f3-bac0-af87484543be) that will backup the data and instance (root drive) [excluding data for `drake-webdav` now] for:

- [drake-license](https://us-east-1.console.aws.amazon.com/backup/home?region=us-east-1#/backupplan/details/83a5c41b-0f95-47f3-bac0-af87484543be/selection/details/15268100-7f19-4826-b99c-c7464a8233da) ~8GB
- [drake-jenkins](https://us-east-1.console.aws.amazon.com/backup/home?region=us-east-1#/backupplan/details/83a5c41b-0f95-47f3-bac0-af87484543be/selection/details/816260dd-0373-4a3d-9ae2-672c2e8addc9) ~1.02 TB (includes data for JENKINS_HOME which we want)
- [drake-cdash](https://us-east-1.console.aws.amazon.com/backup/home?region=us-east-1#/backupplan/details/83a5c41b-0f95-47f3-bac0-af87484543be/selection/details/75733362-5660-4d20-b1e5-102dfc91be27) ~96GB
- [drake-mongo](https://us-east-1.console.aws.amazon.com/backup/home?region=us-east-1#/backupplan/details/83a5c41b-0f95-47f3-bac0-af87484543be/selection/details/32ea906e-337a-4825-a098-80740dc58e23) ~108GB
- [drake-aurora-mysql](https://us-east-1.console.aws.amazon.com/backup/home?region=us-east-1#/backupplan/details/83a5c41b-0f95-47f3-bac0-af87484543be/selection/details/9c8b821c-e188-4f50-aa7a-db624787ed21), no size given? [latest backup job](https://us-east-1.console.aws.amazon.com/backup/home?region=us-east-1#/jobs/backup/045A44B9-FA75-20E4-7AC9-200963BC3A1B)
- [drake-webdav](https://us-east-1.console.aws.amazon.com/backup/home?region=us-east-1#/backupplan/details/83a5c41b-0f95-47f3-bac0-af87484543be/selection/details/2687099a-fcaa-4d64-b028-6903a888e601), just created this, [job that ran last night was [1008 GB](https://us-east-1.console.aws.amazon.com/backup/home?region=us-east-1#/jobs/backup/68262672-D7FC-44AA-ED14-ABDA5FE00EE3) but I will check on the backup size tomorrow to confirm the data volume is excluded this time.

All of these jobs run on the same [DailyBackups schedule](https://us-east-1.console.aws.amazon.com/backup/home?region=us-east-1#/backupplan/details/83a5c41b-0f95-47f3-bac0-af87484543be/version/Y2JiY2Y4NjItZGVjOC00NzliLTgwZGYtMzhkN2JjMWYxMjM2/rules/5c017430-5dab-476a-9a5c-c1f34a45f8da) which currently starts at 5am eastern, and backups are retained for 5 weeks.  Of all of these, `drake-webdav` is really the only one that can be bootstrapped from a clean OS quickly.  We could remove it from the backup plan if desired, but I think it's nice to be able to just restore an instance ASAP from one of these backups, switch over the elastic IP, and call yourself restored.

The complication is that @BetsyMcPhail and I originally desired not to tinker with is the backup rules, and as I've learned through testing the Jenkins setup, this creates more work than we need to.  For example, consider a day when drake-jenkins (the whole instance) is completely broken for unknown reasons.  Then what you do is

1. Pause the old jenkins (`service jenkins stop` or pause the machine via EC2).
2. Restore the drake-jenkins full backup by deploying it to a new AWS instance (can only ever go to a new instance) via their Restore functionality.
3. If (2) is working OK, then delete the backup rules hooked up to (1) and assign a new backup resource that backs up the new instance / volume.

This process will be **significantly** faster than trying to manually restore, **but** we also have restore procedures for just if JENKINS_HOME is broken (TLDR: keep the instance, mount the old data volume and `rsync` it, detach the backup volume and resume jenkins).  So this week I want to show her the other options I've found are and make sure we're all on the same page about how the backups work, and what the drawbacks are of each method.

Updated the description with outcomes (what to document where), just want to discuss with people how the instance change from a restore affects backups (namely... you need to create a new one!) and if we're ok with that as a policy. Changed priority now that things are confirmed backing up / not failing each night.

svenevs, 2024-01-24:
Re-opening to flag to myself to re-check the `drake-webdav` backup rule.  We learned that if you just select the instance it backs up all volumes associated with it, the new backup rule _hopefully_ is going to just backup the instance and the 8GB volume now.

Pending the next backup (tomorrow morning), either document this nuance, or really, stop backing up `drake-webdav` given that [bootstrapping a new one is a few copy-paste commands away](https://github.com/RobotLocomotion/drake-ci/tree/main/cache_server)

svenevs, 2024-01-25:
We cannot get it to only backup the 8GB volume.  We are going to stop backing up `drake-webdav` entirely to avoid wasting storage costs.

closed 2024-01-25 by svenevs