RobotLocomotion/drake #20718: Unknown macOS Sonoma CDash Submission Failures

svenevs, 2023-12-31:
Seems to only happen on sonoma.  I went back through the logs, there _were_ successful submissions in the past. I thought the jenkins changes I was making were affecting this, but it may not actually be related at all, mac-arm-ventura is fine.  It's very unclear to me what's wrong here.

See end of reference build logs:

- (broken sonoma) https://drake-jenkins.csail.mit.edu/job/mac-arm-sonoma-clang-bazel-experimental-release/28/consoleFull
- (fine ventura) https://drake-jenkins.csail.mit.edu/job/mac-arm-ventura-clang-bazel-experimental-release/3810/consoleFull

I have no clue what to do about this, nothing obvious coming out on cdash server logs either.  Not my domain.  I actually have the jenkins credential security looser than we need right now AFAICT, still doesn't fix it.

CC @BetsyMcPhail @jwnimmer-tri

svenevs, 2024-01-02:
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sonoma-clang-bazel-nightly-everything-release/21/consoleFull

```
[5:15:11 AM]  Executed 7677 out of 7677 tests: 7677 tests pass.
[5:15:11 AM]  There were tests whose specified size is too big. Use the --test_verbose_timeout_warnings command line option to see which ones these are.
[5:15:11 AM]  INFO: Build completed successfully, 44627 total actions
[5:15:11 AM]  INFO: Build Event Protocol files produced successfully.
[5:16:08 AM]     Error when uploading file: /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/_bazel_admin/Testing/20240102-0937/Update.xml
[5:16:08 AM]     Error message was: LibreSSL/3.3.6: error:0DFFF0A1:lib(13):func(4095):reason(161)
[5:16:08 AM]     Problems when submitting via HTTP
[5:16:08 AM]  CMake Warning at /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/driver/configurations/bazel/step-build.cmake:121 (message):
[5:16:08 AM]    *** CTest submit update part was not successful
[5:16:08 AM]  Call Stack (most recent call first):
[5:16:08 AM]    /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/driver/functions.cmake:430 (include)
[5:16:08 AM]    /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/driver/configurations/bazel.cmake:307 (execute_step)
[5:16:08 AM]    /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/ctest_driver_script.cmake:129 (include)
[5:16:08 AM]  
[5:16:08 AM]  
[5:17:17 AM]     Error when uploading file: /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/_bazel_admin/Testing/20240102-0937/Upload.xml
[5:17:17 AM]     Error message was: LibreSSL/3.3.6: error:0DFFF0A1:lib(13):func(4095):reason(161)
[5:17:17 AM]     Problems when submitting via HTTP
[5:17:17 AM]  CMake Warning at /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/driver/configurations/bazel/step-build.cmake:135 (message):
[5:17:17 AM]    *** CTest submit upload part was not successful
[5:17:17 AM]  Call Stack (most recent call first):
[5:17:17 AM]    /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/driver/functions.cmake:430 (include)
[5:17:17 AM]    /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/driver/configurations/bazel.cmake:307 (execute_step)
[5:17:17 AM]    /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/ctest_driver_script.cmake:129 (include)
[5:17:17 AM]  
[5:17:17 AM]  
[5:18:15 AM]  Error in HttpRequest
[5:18:15 AM]  CMake Warning at /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/driver/configurations/bazel/step-build.cmake:146 (message):
[5:18:15 AM]    *** CTest submit CDash upload Bazel JSON was not successful
[5:18:15 AM]  Call Stack (most recent call first):
[5:18:15 AM]    /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/driver/functions.cmake:430 (include)
[5:18:15 AM]    /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/driver/configurations/bazel.cmake:307 (execute_step)
[5:18:15 AM]    /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/ctest_driver_script.cmake:129 (include)
[5:18:15 AM]  
[5:18:15 AM]  
[5:19:13 AM]     Error when uploading file: /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/_bazel_admin/Testing/20240102-0937/Done.xml
[5:19:13 AM]     Error message was: LibreSSL/3.3.6: error:0DFFF0A1:lib(13):func(4095):reason(161)
[5:19:13 AM]     Problems when submitting via HTTP
[5:19:13 AM]  CMake Warning at /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/driver/configurations/bazel/step-build.cmake:183 (message):
[5:19:13 AM]    *** CTest submit done part was not successful
[5:19:13 AM]  Call Stack (most recent call first):
[5:19:13 AM]    /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/driver/functions.cmake:430 (include)
[5:19:13 AM]    /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/driver/configurations/bazel.cmake:307 (execute_step)
[5:19:13 AM]    /Users/admin/workspace/mac-arm-sonoma-clang-bazel-nightly-everything-release/ci/ctest_driver_script.cmake:129 (include)
```

Still a problem, discussing internally.

BetsyMcPhail, 2024-01-05:
Possibly related to https://github.com/Homebrew/homebrew-core/issues/158759#issuecomment-1874091015

svenevs, 2024-01-05:
Issue description updated now that we understand the solution better.  https://github.com/RobotLocomotion/drake-ci/pull/261 is a temporary bypass, there's no further action for us to take right now.  In the future, the workaround should be removed when the problem is fixed upstream (in brew).

BetsyMcPhail, 2025-04-07:
@Aiden2244 can you please give this a try on the AWS Sonoma image.  The steps would be to create a new PR to revert https://github.com/RobotLocomotion/drake-ci/pull/261 and then test the PR against  `mac-arm-sonoma-clang-bazel-experimental-release-aws`

BetsyMcPhail, 2025-04-17:
Now that all Mac jobs have been moved to AWS, the revert PR could be merged.

closed 2025-04-17 by BetsyMcPhail