RobotLocomotion/drake #20739: Enable more mac-arm cache, possibly let Nighly populate cache again

svenevs, 2024-01-04:
We have a lot of free space on the mac-arm cache server (typically over 700GB).  We had been discussing adding more.

https://github.com/RobotLocomotion/drake/issues/19341#issuecomment-1602002794

We only let continuous jobs populate the cache currently.  Previously nightly was allowed to.  If we re-enable nightly, then I'd want to cache the debug builds.  Or we may want to just add some debug (or everything) builds to continuous on the mac-arm front instead?  Historically, debug builds are where there are macOS solver tolerance issues, giving me a slight preference toward saying "add a continuous-debug for ventura".  @jwnimmer-tri do you have a preference between adding a new continuous job, or making nightly able to populate the cache again?

Developer note: until `--remote_download_minimal` (ETA bazel 7) is available, debug builds can be slower (too much I/O) so double check that enable caching (on whatever we do) is.

CC @BetsyMcPhail

svenevs, 2024-01-04:
Clarification: it might be, given the other dials we are turning with orka and throughput, smartest to leave this on the backburner.  It may just get closed as #wontfix.

svenevs, 2024-01-11:
The only thing I see possibly worth doing here is that currently we have

https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sonoma-clang-bazel-nightly-debug/

There is no `ventura` equivalent.  This was probably a mistake by yours truly... so either 

1. Add `ventura` nightly equivalent, and add `continuous` flavors, or maybe
2. Delete that `sonoma` `nightly-debug` and do nothing.

AKA things seem to be working fine with 2 or 3 nodes, we can close this as #wontfix, or possibly start doing some debug caching.  When I brought this up, I didn't realize that `sonoma` was the only nightly debug job, I thought we had one for ventura too.  So caching those in continuous (since we likely have enough space) seemed like a win.

BetsyMcPhail, 2024-01-12:
The sonoma nightly debug job was added per https://github.com/RobotLocomotion/drake/issues/20339#issuecomment-1807160642. 

There may have been a typo in that issue as the description does not match the list of jobs "The extra-testing nightlies (debug) should run on the same macOS as the packaging jobs, i.e., the _oldest_ one". It seems like **mac-arm-sonoma-clang-bazel-nightly-debug** should be changed to **mac-arm-ventura-clang-bazel-nightly-debug**

svenevs, 2024-01-19:
@jwnimmer-tri do you agree that the following change should be performed?

> It seems like mac-arm-sonoma-clang-bazel-nightly-debug should be changed to mac-arm-ventura-clang-bazel-nightly-debug

Or, alternatively, we should have enough availability to run both if preferred.  Historically the `everything-debug` jobs have been useful for solver tolerance problems, but it seems like most of those have been fixed to not be tethered to the original drake dev's machine.

jwnimmer-tri, 2024-01-22:
> There may have been a typo in that issue as the description does not match the list of jobs ...

Great catch, I did make a typo.

The `mac-arm-sonoma-clang-bazel-nightly-debug` should be replaced with `mac-arm-ventura-clang-bazel-nightly-debug` (i.e., switch it from Sonoma to Ventura).  That way, we're doing the extra debug-assertion testing on the same OS as our releases.

There's no reason to keep running both (Sonoma).  We'll just turn that one off.

> ... smartest to leave [remote caching] on the backburner.

Agreed.  There's no need to change the remote caching settings for macOS Nightly.  Once we're up on Bazel 7 and we have _linux_ caching more fine-tuned, we could imagine circling back and trying on macOS as well, but let's not worry about that.

---

In short, the only action here is to rename the one build (switch its OS).

closed 2024-01-22 by svenevs