RobotLocomotion/drake #20741: Add crate_universe into new_release

jwnimmer-tri, 2024-01-04:
Related to #20723.

See the TODO here:

https://github.com/RobotLocomotion/drake/blob/bdcd5ff96a1dfed77791302fb499ec2aa831a03e/tools/workspace/crate_universe/README.md#L18-L20

The overall goal is that `new_release` also upgrades the crate packages.  Ideally that's by interfacing with `new_release` somehow, but if that's too difficult then amending the docs at https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/README.md#semi-automated-monthly-upgrades would be OK also.

\CC @BetsyMcPhail FYI

jwnimmer-tri, 2024-01-04:
Maybe to clarify... the existing `upgrade.sh` script in that same directory performs the upgrade, by editing the relevant files.  It doesn't do anything in terms of adding to `git`, nor is able to check whether anything _needs_ an upgrade.

I think the behavior we're looking for here is that as part of the monthly process we _always_ run this `upgrade.sh` and (assuming something changed) we stage and commit that.

jwnimmer-tri, 2024-01-18:
The real outcome we want is that we don't forget to keep running this.

Instead of changing `new_release`, maybe instead we should make an issue template for the monthly task, and the template can have a checkbox to remind us to run it.  The template should prompt to encourage itemizing the PR numbers(s).

jwnimmer-tri, 2024-01-18:
Another idea: the issue template could remind us to check for a newer bazel version (or, we could have new_release automate that check).

jwnimmer-tri, 2024-01-25:
Remaining work: make a nice issue template to simplify our lives.  Checklists rule!

BetsyMcPhail, 2024-07-09:
Reading through the history of this issue, the only task left is to create a template with the for the monthly update issue. The template should follow the same pattern as previous issues and include a checkbox with a reminder to check for Bazel updates.

After completing the last few monthly upgrades, my workflow is to:
- Create the issue by copying and pasting the previous issue, with the "previous issue" link and month updated.
- Run the `new_release `script to get a list of packages to update
- Work through the list creating PRs linked back to the issue

For me, adding Bazel to the list of packages to update, even if that update is then done by hand, would be the most helpful. I have a simple PR to do that, #21697

jwnimmer-tri, 2024-07-11:
We've improved some things, and think this is working well enough now.

closed 2024-07-11 by jwnimmer-tri