RobotLocomotion/drake #20754: `common.schema.Transform.Sample()` silently discards `base_frame`

ggould-tri, 2024-01-08:
### What happened?

The `common.schema.Transform` class has a member `base_frame`.

The `common.schema.Transform.Sample()` method returns `RigidTransformd`, not `Transform`.

`math.RigidTransformd` does not have a `base_frame`, so anyone consuming a `Transform` via `Sample` has to save off the base frame out-of-line.

It is unintuitive that `Sample` discards non-distribution information; moreover because `RigidTransformd` convert-contructs to `common.schema.Transform`, the extremely obvious code
```
Transform transform_distribution = [ mumble ]
Transform concrete = transform_distribution.Sample()
```
...compiles and runs without warning but is incorrect.

Assuming that we don't want `common.schema.Transform.Sample()` to return `common.schema.Transform`, it should instead assert that `base_frame` is nullopt.

(cc @calderpg-tri per our discussion)

jwnimmer-tri, 2024-01-08:
> Assuming that we don't want common.schema.Transform.Sample() to return common.schema.Transform, it should instead assert that base_frame is nullopt.

This would be fine by me.

If we like, we could add a different function name to return a sampled copy of the `Transform` with the base name intact.

closed 2024-01-11 by ggould-tri

jwnimmer-tri, 2024-01-13:
I would have thought that the same complaint about "silently ignoring base_frame" would have applied to `GetDeterministicValue()` as well?

PR #20756 (2024-01-11): Throw when Transform::Sample is called with a base frame.
 * Resolves #20754:  Sample() was silently discarding base_frame.
 * Throws when called in the dangerous case.
 * Adds new SampleAsTransform() method that samples into the same type, preseving the frame.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20756)
commit 347b6940 (2024-01-11): Throw when Transform::Sample is called with a base frame. (#20756)

 * Resolves #20754:  Sample() was silently discarding base_frame.
 * Throws when called in the dangerous case.
 * Adds new SampleAsTransform() method that samples into the same
   type, preseving the frame.