RobotLocomotion/drake #20767: Clean up VTK OpenGL abort() calls

svenevs, 2024-01-11:
[`vtkXOpenGLRenderWindow.cxx`](https://gitlab.kitware.com/vtk/vtk/-/blob/master/Rendering/OpenGL2/vtkXOpenGLRenderWindow.cxx) has a handful of `abort()` calls, drake bypasses them by hooking into `vtkErrorMacro` which happens to always precede the `abort()`.

VTK should remove these, in favor of more appropriate error reporting.  Will require a brief discussion with vtk community to determine impact / how error reporting should be performed.

Drake just adds observers

https://github.com/RobotLocomotion/drake/blob/ed2cc4cdfa8314073ddb09551afba804beeaec80/geometry/render_vtk/internal_render_engine_vtk.cc#L632-L636

Offshoot of #20455.

CC @jwnimmer-tri @BetsyMcPhail @kwryankrattiger @mathstuf

jwnimmer-tri, 2024-01-16:
> ... drake bypasses them by hooking into vtkErrorMacro which happens to always precede the abort().

Probably obvious but might be worth me clarifying -- Drake adds an observer for `vtkCommand::ErrorEvent`, and in that observer _throws an exception_ to accomplish a non-local exit that avoids the `abort()`.  Luckily, the VTK code nearby happens to be sufficiently exception-safe that we can get away with this.

BetsyMcPhail, 2024-04-11:
MR has been closed (not implemented) in VTK

closed 2024-04-11 by BetsyMcPhail