RobotLocomotion/drake #20792: Coverage results are not being upload to CDash

BetsyMcPhail, 2024-01-18:
### What happened?

Results from coverage jobs (e.g. https://drake-jenkins.csail.mit.edu/view/Coverage/job/linux-jammy-gcc-bazel-nightly-coverage/100/console) are no longer being uploaded to CDash.

The last time results were uploaded was October 4, 2023

https://drake-cdash.csail.mit.edu/index.php?project=Drake&begin=2023-10-01&end=2024-01-17&filtercount=1&showfilters=1&field1=hascoverage&compare1=1&value1=1

### Relevant log output

```shell
[8:34:08 AM]  -- Submitted to CDash with build id 1757226
[8:34:09 AM]  Traceback (most recent call last):
[8:34:09 AM]    File "/media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-nightly-coverage/src/tools/dynamic_analysis/kcov_tool", line 133, in <module>
[8:34:09 AM]      main()
[8:34:09 AM]    File "/media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-nightly-coverage/src/tools/dynamic_analysis/kcov_tool", line 129, in main
[8:34:09 AM]      args.func(args)
[8:34:09 AM]    File "/media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-nightly-coverage/src/tools/dynamic_analysis/kcov_tool", line 96, in ci_merge
[8:34:09 AM]      do_merge(os.path.join(WORKSPACE, "bazel-kcov"))
[8:34:09 AM]    File "/media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-nightly-coverage/src/tools/dynamic_analysis/kcov_tool", line 87, in do_merge
[8:34:09 AM]      maybe_expand_zipped_kcov_data(d)
[8:34:09 AM]    File "/media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-nightly-coverage/src/tools/dynamic_analysis/kcov_tool", line 46, in maybe_expand_zipped_kcov_data
[8:34:09 AM]      with zipfile.ZipFile(zip_path, 'r') as zf:
[8:34:09 AM]    File "/usr/lib/python3.10/zipfile.py", line 1251, in __init__
[8:34:09 AM]      self.fp = io.open(file, filemode)
[8:34:09 AM]  FileNotFoundError: [Errno 2] No such file or directory: '/media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-nightly-coverage/src/bazel-testlogs/tools/install/libdrake/py/exported_symbols_test/test.outputs/outputs.zip'
[8:34:09 AM]  Error copying file "/media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-nightly-coverage/src/bazel-kcov/kcov-merged/cobertura.xml" to "/media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-nightly-coverage/src/bazel-kcov/kcov-merged/coverage.xml".
```

jwnimmer-tri, 2024-01-22:
`kcov_tool` is crashing on `bazel-testlogs/tools/install/libdrake/py/exported_symbols_test/test.outputs/outputs.zip`.

BetsyMcPhail, 2024-01-23:
I confirmed that coverage results were uploaded last night. The table will continue to fill in when the weekly jobs run this weekend.

[image]

closed 2024-01-23 by BetsyMcPhail