RobotLocomotion/drake #20795: Consolidate drake-jenkins documentation

svenevs, 2024-01-19:
There are some sections described in the uber CI doc that need to get merged with the new drake-jenkins doc.

1. @svenevs: do the doc merge.
2. @svenevs: see TODO in `drake-jenkins` doc, go find the notes you have on the `keytool` drama that broke MacStadium.  Nobody should have to relearn that.
3. @BetsyMcPhail: review the new drake-jenkins doc, make sure nothing is missing from the uber doc, and delete the now redundant data from the uber CI doc and link to the `drake-jenkins` doc instead.

CC @jwnimmer-tri FYI.  I want to do this with fresh eyes... 95% of the jenkins docs are done now though.

svenevs, 2024-01-24:
Alrighty, @BetsyMcPhail when you are available, please review the new `drake-jenkins` document.  I have gone through the sections describing Jenkins in the original document, there's two sections (one at the top, and one near the bottom) and imported them.  In the original document, they are highlighted in yellow with comments explaining that I think they can be deleted, but my hope is that by leaving the original intact for now it will be helpful for you to review.

The new document also has comprehensive documentation on everything that was the jenkins update saga, including directions for monthly updates as well as distribution upgrades.  After a non-me human has reviewed the document, I think we're ready to 

1. Delete the sections from the original document.
2. Close this ticket :partying_face:

BetsyMcPhail, 2024-01-25:
I confirmed all of the relevant information was moved to the new document. The new document is looking good, I can't think of anything it's missing at this time.

closed 2024-01-25 by BetsyMcPhail