RobotLocomotion/drake #20809: Packages and classes missing

julialopezgomez, 2024-01-23:
I am trying to reimplement the code for the C-IRIS algorithm, with the Drake implementation provided in this paper: Certified Polyhedral Decompositions of Collision-Free Configuration Space (https://arxiv.org/abs/2302.12219), link to code here: (https://deepnote.com/workspace/alexandre-amice-c018b305-0386-4703-9474-01b867e6efea/project/C-IRIS-7e82e4f5-f47a-475a-aad3-c88093ed36c6/notebook/2d_example_bilinear_alternation-14f1ee8c795e499ca7f577b6885c10e9). 

When trying to import packages and classes, some are not found, in specific:
- The class: 'IrisInRationalConfigurationSpace' from 'pydrake.geometry.optimization'
- The entire module: 'pydrake.geometry.optimization_dev'

They work in deepnote, but I have tried several drake versions locally and none work. Is there a way in which I can get a version - or any solution - to use these implementations in my local computer?

Operating system: macOS Sonoma 14.2.1 (it is an intel computer, not arm64, is that a problem?)
Installation method: pip
Using python 3.11 with macOS homebrew distribution
Drake version: last nightly release 'drake-0.0.20240123', and last stable version tried too 'drake-1.25.0'

AlexandreAmice, 2024-01-23:
Thank you for your interest in our work. 

If you would like the exact same branch that is used in the Deepnote notebooks it is available on [my own fork](https://github.com/AlexandreAmice/drake/tree/C_Iris_Wafr_Journal_Examples). It will require you to build [Drake from source](https://drake.mit.edu/from_source.html) with Mosek enabled. The other option is to run you code in the same [Docker container](https://hub.docker.com/layers/amice/c_iris/deepnote_v0/images/sha256-6f237cc3d4ed9ec45c59edb64f5c2f8db19f1786430e7232d19de0c8c18e532c?context=repo) that is used for the deepnote notebooks

Please note that the vast majority of the functionality for using C-IRIS is available in the master branch of Drake that you can install via pip. The only thing that is missing would be `IrisInRationalConfigurationSpace` which we can look into merging into master.

The code for C-IRIS itself is in the `pydrake.geometry.optimization` module. The [documentation is here](https://drake.mit.edu/pydrake/pydrake.geometry.optimization.html#pydrake.geometry.optimization.CspaceFreePolytope). The deepnote notebooks are based on a development branch of drake hence the different module name.

Please let me know if you have any more questions.

hongkai-dai, 2024-01-24:
As @AlexandreAmice has mentioned, most of the python binding for C-IRIS is in the master version of pydrake already. Instead of using `pydrake.geometry.optimization_dev`, I think you can find a lot of the functions in `pydrake.geometry.optimization`.

BTW, the source code is all written in C++. So if you want to reproduce the code, you could refer to our C++ implementation https://github.com/RobotLocomotion/drake/blob/master/geometry/optimization/cspace_free_polytope.h#L44

closed 2024-02-21 by AlexandreAmice