RobotLocomotion/drake #20835: Early CI support for prototyping Noble

jwnimmer-tri, 2024-01-29:
Over the next few months, we'll want to start preparing for Ubuntu 24.04 ("Noble") so that we can officially support it once it's released.  To that end, it would help to be able to test some early PRs in CI.

Please create a job `linux-noble-unprovisioned-gcc-bazel-experimental-release` so that PR authors can launch that.  We don't need any Continuous, Nightly, nor Provisioned flavors for now.

It's possible that the the base image we start from isn't available yet, so this ticket isn't immediately actionable.

jwnimmer-tri, 2024-03-16:
See also #21008.

jwnimmer-tri, 2024-04-25:
As of today (or yesterday?) there are official EC2 AMIs now.

closed 2024-05-03 by BetsyMcPhail

BetsyMcPhail, 2024-05-03:
`linux-noble-unprovisioned-gcc-bazel-experimental-release` is ready for testing. Not surprisingly there's a few test failures, but CI support (and this ticket) are complete.