RobotLocomotion/drake #20839: HPolyhedron::MaximumVolumeInscribedEllipsoid Failure with Mosek

cohnt, 2024-01-29:
### What happened?

I encountered the [following HPolyhedron](https://gist.github.com/cohnt/56e887560451751d88149b1ff42981d4) while running a variant of IrisInConfigurationSpace. When computing the maximum volume inscribed ellipsoid, Mosek fails, causing IRIS itself to fail. The cause is unclear -- Mosek reports a solver-specific error. Creating the issue to see if anyone else has encountered it before or has insights (maybe @hongkai-dai?), but I'll be working on debugging this one myself.

### Version

20240122164254 46cbb1316da2f71f64b0a1c4f083533932b12297

### What operating system are you using?

Unknown / Other, Ubuntu 22.04

### What installation option are you using?

compiled from source code using CMake

cohnt, 2024-01-29:
Per @AlexandreAmice's suggestion, normalizing the rows of the matrix `[A, b]` fixes the issue. I'll plan to implement this normalizing in HPolyhedron, to avoid the issue in the future.

It's also probably worth adding an example HPolyhedron with poor numerics as a test case. This example is probably much larger than a MWE, so I'll work on shrinking it down.

AlexandreAmice, 2024-01-30:
Since this came up during IRIS, I think it's worth outputting normalized rows for all the IRIS hyperplanes.

closed 2024-02-11 by rpoyner-tri