RobotLocomotion/drake #20841: Joint naming inconsistency in Drake API

sherm1, 2024-01-30:
(Capturing review discussion from #20827)

We have BallRpyJoint, QuaternionFloatingJoint, and RpyFloatingJoint (per #14949). Likely we'll have BallQuaternionJoint in the future.

The ideal names IMO would have been BallRpy, BallQuaternion, FloatingRpy, and FloatingQuaternion because then the joint variants would alphabetize together. We could move that way by renaming the existing floating joint to floating_quaternion_joint. But is this really going to solve any user problems? I'm guessing that people are able to find the joints they need. And the floating joints are most often added automatically under the covers without the user needing to name them.

Ideally we would name all the Drake joints consistently. However, my feeling now is that it isn't really worth the pain to rename existing joints. It may be enough to have Ball{Rpy,Quaternion}Joint and {Rpy,Quaternion}FloatingJoint.

I'm leaning towards "won't fix", but this is open for discussion.

jwnimmer-tri, 2024-01-30:
I agree with "won't fix".

amcastro-tri, 2024-02-05:
The other option not mentioned here is to go from `BallRpyJoint` to `RpyBallJoint` so that all names are consistent. I agree this low priority though and I'd be ok with closing.