RobotLocomotion/drake #20847: //solvers:mosek_solver_internal_test: Segmentation fault

liangfok, 2024-01-30:
### What happened?

The following failure occured in nightly production:

```
/media/ephemeral0/ubuntu/workspace/linux-jammy-clang-bazel-nightly-coverage/_bazel_ubuntu/ca8cca4382e64e0e68b3df6a056070b5/execroot/drake/bazel-out/k8-dbg/bin/solvers/mosek_solver_internal_test.runfiles/drake/tools/dynamic_analysis/kcov: line 65: 75308 Segmentation fault      (core dumped) kcov "--include-path=/media/ephemeral0/ubuntu/workspace/linux-jammy-clang-bazel-nightly-coverage/src" --verify --python-parse=python3 --exclude-pattern=third_party "--replace-src-path=/proc/self/cwd:/media/ephemeral0/ubuntu/workspace/linux-jammy-clang-bazel-nightly-coverage/src" "/media/ephemeral0/ubuntu/workspace/linux-jammy-clang-bazel-nightly-coverage/_bazel_ubuntu/ca8cca4382e64e0e68b3df6a056070b5/execroot/drake/bazel-out/k8-dbg/testlogs/solvers/mosek_solver_internal_test/test.outputs/kcov" solvers/mosek_solver_internal_test
Could not create "/media/ephemeral0/ubuntu/workspace/linux-jammy-clang-bazel-nightly-coverage/_bazel_ubuntu/ca8cca4382e64e0e68b3df6a056070b5/execroot/drake/bazel-out/k8-dbg/testlogs/solvers/mosek_solver_internal_test/test.outputs/outputs.zip": zip not found or failed

//solvers:mosek_solver_internal_test                                     FAILED in 3.0s
```

Links:
1. https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-clang-bazel-nightly-coverage/116/
2. https://drake-cdash.csail.mit.edu/test/1343352056

liangfok, 2024-01-30:
Closing since this is first occurrence.

SeanCurtis-TRI, 2024-03-04:
We had another incidence today -- albeit in a different test.

https://drake-jenkins.csail.mit.edu/view/Weekly%20Production/job/linux-jammy-clang-bazel-weekly-everything-coverage/21/

BetsyMcPhail, 2024-03-25:
Three more failures, all in different tests:
- https://drake-jenkins.csail.mit.edu/view/Weekly%20Production/job/linux-jammy-clang-bazel-weekly-everything-coverage/25/
- https://drake-jenkins.csail.mit.edu/view/Weekly%20Production/job/linux-jammy-clang-bazel-weekly-mosek-coverage/26/
- https://drake-jenkins.csail.mit.edu/view/Weekly%20Production/job/linux-jammy-clang-bazel-weekly-snopt-coverage/24/console

BetsyMcPhail, 2024-03-29:
3/29 segfault in `//multibody/contact_solvers:minimum_degree_ordering_test`  in [https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-clang-bazel-nightly-coverage/175/](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-clang-bazel-nightly-coverage/175/)

BetsyMcPhail, 2024-04-01:
4/1 segfault in `//math:hopf_coordinate_test`: [linux-jammy-gcc-bazel-nightly-coverage](https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-nightly-coverage/176/)

xuchenhan-tri, 2024-04-02:
Reopening the issue per the buildcop playbook. I tried reproducing the segfault locally but wasn't able to do it with a handful of runs.

BetsyMcPhail, 2024-04-09:
4/9 `//multibody/fem:dirichlet_boundary_condition_test` [linux-jammy-clang-bazel-nightly-coverage](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-clang-bazel-nightly-coverage/186/console)

BetsyMcPhail, 2024-04-15:
4/15: `//multibody/parsing:package_map_remote_test` https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-clang-bazel-nightly-coverage/192/

BetsyMcPhail, 2024-05-20:
5/20: [//common:text_logging_test](https://drake-cdash.csail.mit.edu/test/1468631004) failed in [linux-jammy-gcc-bazel-weekly-everything-coverage](https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-weekly-everything-coverage/33/)

BetsyMcPhail, 2024-05-21:
5/21 //common:dummy_value_test failed in  linux-jammy-clang-bazel-nightly-coverage

williamjallen, 2024-09-19:
9/19 //common:extract_double_test failed in [linux-jammy-clang-bazel-nightly-coverage](https://drake-jenkins.csail.mit.edu/view/Nightly/job/linux-jammy-clang-bazel-nightly-coverage/350/consoleFull)

williamjallen, 2024-09-24:
9/24 //manipulation/kuka_iiwa:iiwa_status_receiver_test failed in https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-nightly-coverage/355/consoleFull

BetsyMcPhail, 2024-11-19:
11/19 - [linux-jammy-clang-bazel-nightly-coverage](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-clang-bazel-nightly-coverage/412/) failed with a segfault in //systems/primitives:vector_log_test

BetsyMcPhail, 2025-01-03:
1/3/25  - `//common:hash_test` failed in [linux-jammy-clang-bazel-nightly-coverage](https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-nightly-coverage/457)

SeanCurtis-TRI, 2025-01-30:
1/30/25 - `//multibody/fem:linear_constitutive_model_data_test` failed in [linux-jammy-gcc-bazel-nightly-coverage](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-gcc-bazel-nightly-coverage/488/)

BetsyMcPhail, 2025-02-05:
2/5 `//geometry:geometry_set_test` failed in https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-gcc-bazel-nightly-coverage/496

BetsyMcPhail, 2025-04-14:
4/14 //common/test_utilities:eigen_matrix_compare_test failed in https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-clang-bazel-nightly-coverage/558/

BetsyMcPhail, 2025-06-06:
6/6 https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-nightly-coverage/621