RobotLocomotion/drake #20848: [website] Wrong auto-scrolling in in pydrake API reference anchors

jwnimmer-tri, 2024-01-30:
Sometimes when clicking a link to one of the pydrake API pages, the browser viewport scrolls to a few lines _after_ the start of the docs that the user wanted.  This is annoying.

I think it's some kind of race condition, so it's not necessarily 100% reproducible, but the following repros for me pretty reliably:

- Open https://drake.mit.edu/pydrake/pydrake.geometry.html#pydrake.geometry.DepthRenderCamera.
- The function docs for `depth_range(...)` have a hyperlink to `pydrake.geometry.DepthRange`.
- Click on the link to `DepthRange`.

jwnimmer-tri, 2026-09-10:
FYI @tyler-yankee this might be worth re-testing after all of our pending Sphinx upgrades are finished.

tyler-yankee, 2026-09-11:
We can check it with newer Sphinx, but my guess would be something to do with the padding/size of the header on the page causing the scroll to mis-locate the position of the anchor?

mwoehlke-kitware, 2026-09-14:
This feels like it might be a browser bug. The right course of action here might be to report it upstream.

What browser are you using? I am not seeing this in Firefox.

jwnimmer-tri, 2026-09-15:
> What browser are you using? I am not seeing this in Firefox.

Firefox on Ubuntu, consistent through many versions over the years.

(Note that per OP, it's not 100% reprodicuble every time, though on retrying this morning I did hit it 3x in a row on the OP link.)

> This feels like it might be a browser bug. The right course of action here might be to report it upstream.

We have a (very) custom style sheet that might be interfering.  Not clear that it's actually an upstream issue.

mwoehlke-kitware, 2026-09-15:
>  Not clear that it's actually an upstream issue.

Possible. But anyway, I couldn't reproduce on Firefox (distro package) on Fedora. I guess I won't be any help on this, at least...

tyler-yankee, 2026-09-15:
As a data point, I can reproduce very reliably on Chrome on both Windows and Ubuntu.

mwoehlke-kitware, 2026-09-15:
Huh. Can reproduce on Falkon. Just opening the reproducer link triggers it.

This... sort of feels like some element's calculated size is initially too *large*. Too *small* would be not strange, but too large?

I think I would start by testing if this can be reproduced from a locally saved page. That way, it's easy to remove or edit `.css` files.

mwoehlke-kitware, 2026-09-15:
...and I can't reproduce with locally-saved content. That *does* suggest it's related to resource loading, and probably will depend on network factors.

jwnimmer-tri, 2026-10-12:
Update -- Kitware contact has been able to reproduce this; time permitting, they will work on root cause and how to fix.

closed 2026-10-21 by SeanCurtis-TRI