RobotLocomotion/drake #20868: Upgrade Externals February 2024

BetsyMcPhail, 2024-02-01:
Follow [Semi-automated monthly upgrades](https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/README.md#semi-automated-monthly-upgrades) documentation.

Prior issue (January 2024): #20723

mwoehlke-kitware, 2024-02-07:
Not too bad this month. There is a glitch with `rules_python`, and VTK needed a patch rebased (but has subsequently been folded into the main PR). The others are split a) due to manual testing requirements that I was only partly able to complete (meshcat), and b) due to non-trivial changes (rules_rust / rust_toolchain got some fiddling to make the upgrade easier going forward (now that we support "scripted" updates :tada:).

mwoehlke-kitware, 2024-02-13:
All PRs have landed.

closed 2024-02-13 by mwoehlke-kitware