RobotLocomotion/drake #20874: Upstream VTK glTF patch

svenevs, 2024-02-02:
Outcropping of #20867.  The patch to be upstreamed:

https://github.com/RobotLocomotion/drake/blob/75ccf854cb3885df189c40e3641798a1d66e16b7/tools/workspace/vtk_internal/patches/gltf_parser.patch

CC @jwnimmer-tri @BetsyMcPhail

jwnimmer-tri, 2024-05-09:
Okay, I've finished the patch inventory @BetsyMcPhail .  We have 4 items to upstream.

BetsyMcPhail, 2024-05-16:
@kwryankrattiger is going to work on this issue.

kwryankrattiger, 2024-05-16:
Patches MR'd to VTK upstream

- [ ] vtkCamera Copy [VTK/VTK!11125](https://gitlab.kitware.com/vtk/vtk/-/merge_requests/11125)
- [ ] Illumination [VTK/VTK!11146](https://gitlab.kitware.com/vtk/vtk/-/merge_requests/11146)
- [ ] GLTF Parser [VTK/VTK!11126](https://gitlab.kitware.com/vtk/vtk/-/merge_requests/11126)
- [ ] GLTF Error Handling [VTK/VTK!11130](https://gitlab.kitware.com/vtk/vtk/-/merge_requests/11130)

BetsyMcPhail, 2024-05-16:
- [ ] Adding https://gitlab.kitware.com/vtk/vtk/-/merge_requests/11117 here as well so it doesn't get forgotten about. It was patched to drake in #21387

kwryankrattiger, 2024-08-07:
I pulled out the configurable exponent part of the illumination patch into a separate MR here: [VTK/VTK!11374](https://gitlab.kitware.com/vtk/vtk/-/merge_requests/11374)

jwnimmer-tri, 2024-10-16:
FYI #22052 should help make the tracking of upstreamable patches easier.

BetsyMcPhail, 2025-02-06:
Upstreamed patches are removed in #22597

BetsyMcPhail, 2025-02-11:
All patches have been upstreamed, the remaining test issues can be tracked in the PR listed above.

closed 2025-02-11 by BetsyMcPhail