RobotLocomotion/drake #20880: meshcat default camera pose in VisualizationConfig

dmcconachie-tri, 2024-02-05:
**Is your feature request related to a problem? Please describe.**
When meshcat re-initializes the camera pose of the rendered window is reset. It would be great if we could close the gap between meldis and meshcat with regards to the ablity to retain the user's desired camera pose.

**Describe the solution you'd like**
A 2 part solution:
- A configuration option as part of `VisualizationConfig` and associated functions that allows the user to set a desired camera pose/focal point/zoom level (i.e.; whatever one changes when moving the camera around and zooming in and out).
- A readout as part of the meshcat controls panel containing a copy-pasteable configuration block

One workflow could then be:
1. Setup a scene for the first time
2. Manually move the camera to a desired pose
3. Copy the configuration stanza out of meshcat into YAML/`VisualizationConfig` initializer.

**Describe alternatives you've considered**
I believe it is programmatically possible to setup the default camera pose; presumably a similar effect could be achieved by manually setting that configuration as part of the user program.

**Additional context**
N/A

jwnimmer-tri, 2024-02-05:
The problem statement was this:
> When meshcat re-initializes the camera pose of the rendered window is reset. It would be great if we could close the gap between meldis and meshcat with regards to the ability to retain the user's desired camera pose.

In your list of solutions, you didn't mention #19183.  Is that a suitable option?

Having a particular window retain its pose is a different thing than having all windows which connect have their pose be initialized to a particular scenario-specific pose.

dmcconachie-tri, 2024-02-05:
I think #19183 would be a superior solution for me personally (my search did not find that issue). The reason it would be superior for me personally is because I don't switch scenes often. If one is frequently switching scenes I could see the `VisualizationConfig` option being superior.

SeanCurtis-TRI, 2024-03-28:
Also, remember, when setting positions directly via viewer properties (as shown above), the positions are expressed in a right-handed y-up world (and not Drake's z-up world).