RobotLocomotion/drake #20891: GcsTrajectoryOptimization: fix the sequence of regions

sadraddini, 2024-02-06:
**Is your feature request related to a problem? Please describe.**
`GcsTrajectoryOptimization::SolvePath` may be too slow using full-blown GCS. It is often much faster to restrict the regions path into a fixed sequence of regions.

I have written tools for this and want to get early feedback and discussion before making a pull request.

**Describe the solution you'd like**
I have two functions. I'm sharing the header here:
```
/** Returns the sequence of regions from a mathematical program result that is
 typically obtained from SolvePath().
 
  @param source specifies the source subgraph. Must have been created from a
  call to AddRegions() on this object, not some other optimization program.
  @param target specifies the target subgraph. Must have been created from a
  call to AddRegions() on this object, not some other optimization program.
 @param tolerance is the tolerance used to determine to select the edges from
 the flow solution resuts that are considered to be part of the path.
 @param include_ends determines whether the start region (belonging to source)
 and end region (belonging to target) are included in the returned sequence. */
  geometry::optimization::ConvexSets GetRegionsPath(
      const Subgraph& source, const Subgraph& target,
      const solvers::MathematicalProgramResult& result,
      const double tolerance = 1e-3, bool include_ends = true) const;
```
And a solve function as follows:
```
  /** Formulates and solves a path on a path graph of convex sets.
  @see geometry::optimization::GraphOfConvexSets::SolveConvexRestriction().
  The path essentially traverses the sequence of convex sets in the
    order they are provided.
  @param convex_set_sequence is the sequence of convex sets to connect.
  @param order is the order of the Bézier curve.
  @param continuous_revolute_joints is a list of indices corresponding to
    continuous revolute joints. @see continuous_revolute_joints().
  @param continuity_order is the order of the continuity constraint. @see
  AddPathContinuityConstraints().
    @param time_cost is the relative weight of the time cost. @see
  AddTimeCost().
    @param path_length_cost is the relative weight of the path length cost. @see
  AddPathLengthCost().
    @param velocity_bounds is the relative weight of the velocity bounds. @see
  AddVelocityBounds().
    @param options include all settings for solving the shortest path problem.
  @see geometry::optimization::GraphOfConvexSetsOptions for further details.
  @returns a CompositeTrajectory of the solution trajectory.

    @throws std::exception if any two consecutive convex sets in the sequence
    do not intersect.
  */
  static trajectories::CompositeTrajectory<double>
  SolvePathViaConvexRestriction(
      const geometry::optimization::ConvexSets& convex_set_sequence,
      const int order,
      std::vector<int> continuous_revolute_joints = std::vector<int>(),
      const std::optional<int> continuity_order = std::nullopt,
      const std::optional<double> time_cost = std::nullopt,
      const std::optional<Eigen::MatrixXd> path_length_cost = std::nullopt,
      const std::optional<std::pair<Eigen::VectorXd, Eigen::VectorXd>>
          velocity_bounds = std::nullopt,
      const geometry::optimization::GraphOfConvexSetsOptions& options = {});
```
What is special about `SolvePathViaConvexRestriction` is that it uses `gcs.graph_of_convex_sets().SolveConvexRestriction`, which often belongs to a simple optimization class, to solve the planning problem. It is orders of magnitude faster than `SolvePath`.

**Describe alternatives you've considered**
* An alternative to `SolvePathViaConvexRestriction` is to wait for fast path planning https://github.com/RobotLocomotion/drake/issues/20388
* An alternative to `GetRegionsPath` is giving access to the underlying vertices and edges in the Gcs. I have talked about it in https://github.com/RobotLocomotion/drake/issues/20186

Want to see what people think @RussTedrake @cohnt @wrangelvid

wrangel-bdai, 2024-02-06:
Alternatively the subgraphs and edges between subgraphs objects could return the edges and vertices. Then we could offer GcsTrajectoryOptimization::SolvePathViaConvexRestriction(edges, options) to extract the trajectory from the result of the mathematical program. 

I am worried that your suggested signature would make it hard to leverage the existing code to employ different velocity limits, time offsets etc. along the path.

sadraddini, 2024-02-06:
> I am worried that your suggested signature would make it hard to leverage the existing code to employ different velocity limits, time offsets etc. along the path.

That's a fair advantage but requires more careful coding. a) It is not straightforward to get edges from regions and we lack the API here. b) A naive trajectory optimization user may not even be interested in the GCS aspect of the class. c) A static method on "give me a trajectory that goes through this vector of regions is simpler. d) When a GCS is used to solve multiple plans, one has to remove source/targets (typically `Point` objects) to avoid having a huge graph after few iterations.

wrangelvid, 2024-02-06:
> That's a fair advantage but requires more careful coding. a) It is not straightforward to get edges from regions and we lack the API here. b) A naive trajectory optimization user may not even be [...]

Here is a draft PR #20895. Let me know what you think.
It took me some time to realize, but I completely agree with you on a) that using the edges is not intuitive at all since the vertices are often connected behind the scenes. So I suggest using the vertices instead.
We can follow up with the ability to remove subgraphs and introduce a wrapper that takes in start and end positions.

The static method to yield a path through a set of regions might require a little bit more care, but could be efficiently written as a wrapper as well to my proposed function with minimal effort.

sadraddini, 2024-02-07:
Continuing the conversation here (not on the draft PR yet):

In your example, you specify active vertices as all vertices in a subgraph. In many examples, we want to be more specific. Therefore, need a `region_to_vertex` map or something of that ilk.

wrangelvid, 2024-02-07:
> Continuing the conversation here (not on the draft PR yet):
> 
> 
> 
> In your example, you specify active vertices as all vertices in a subgraph. In many examples, we want to be more specific. Therefore, need a `region_to_vertex` map or something of that ilk. 

Good point! The order of vertices matches the order of regions, so one could use that. But I like the way you think, perhaps we could add another constructor to AddRegions with a map of region name -> convex set. Then we could return a map of region name -> vertex in Vertices()

wrangelvid, 2024-02-08:
@jwnimmer-tri do you have any thoughts on returning a list of vertices or a dictionary that maps assigned region names to vertices?

wrangelvid, 2024-02-16:
Interesting. Generally the GCS machinery should be treated independent of the source of the sets, so an IrisRegion type/struct may not make a lot of sense on drake master.

wrangelvid, 2024-02-19:
#20895 should resolve this feature request ✅

closed 2024-02-19 by wrangelvid