RobotLocomotion/drake #20907: Add/ improve iiwa7 velocity and acceleration limits

nepfaff, 2024-02-08:
**Is your feature request related to a problem? Please describe.**

The iiwa14 limits have been updated as tracked by [this issue](https://github.com/RobotLocomotion/drake/issues/15613). However, the iiwa7 models don't yet have acceleration limits.
The iiwa7 velocity limits seem large at 10 rad/s based on the much smaller iiwa14 limits. I have not yet tested this myself, so the current velocity limits might be reasonable.

I want to use the models to design trajectories at the velocity and acceleration limits of the iiwa. Hence, more realistic limits would be useful in order not to damage the robot.

**Describe the solution you'd like**
The iiwa14 acceleration limits were obtained from a master's thesis (see [here](https://github.com/RobotLocomotion/drake/pull/16377)). The iiwa7 limits are probably unavailable and would have to be determined using additional experiments.

Would it be reasonable to copy the iiwa14 limits for the iiwa7? They won't be completely accurate, but maybe it will be a conservative estimate much better than the current values/ missing values.

nepfaff, 2024-02-09:
For reference, I started working on this and should have a PR ready soon

nepfaff, 2024-02-09:
This PR corrects the effort and velocity limits: https://github.com/RobotLocomotion/drake/pull/20917

I will create a separate PR for the acceleration limits as these will be an estimate rather than from some documentation.

jwnimmer-tri, 2024-02-13:
> For reference, I started working on this and should have a PR ready soon.

Thanks!  I've assigned this issue to you for tracking / clarity purposes in that case.

nepfaff, 2024-02-13:
Closing this as both PRs are merged

closed 2024-02-13 by nepfaff