RobotLocomotion/drake #20936: ApplyCameraConfig uses the fps setting to set the Lcm publishing rate, but does not create a sampled-data camera.

RussTedrake, 2024-02-13:
This is fine for the hardware_station workflow using LCM, but is potentially confusing/insufficient for the in-process simulation workflow.

Related to #20055.
cc @IanTheEngineer , @dmcconachie-tri , @adamw8 @pangtao22

jwnimmer-tri, 2024-02-13:
This is extremely difficult to solve correctly until #20256 is fixed.  The sampling _must_ be done by the System that publishes the data, or else it will be wrong by 1 frame.

If you are only ever cascading the image output into another input port (and never publishing it), the work-around is that after calling ApplyCameraConfig you can insert a ZOH yourself, inbetween the newly-added continuous-time sensor and whatever downstream receiver you _know for a fact_ does not try to publish mis-timed images.

jwnimmer-tri, 2024-02-13:
It would be fair to say that the documentation is not sufficiently clear about what is currently does.  (As Grant says: if it confuses it is confusing.)  If it helps, I can take a stab at clarifying our documentation of the current behavior.