RobotLocomotion/drake #20941: Meshcat support for arbitrary object_json

jwnimmer-tri, 2024-02-14:
**Is your feature request related to a problem? Please describe.**

Meshcat uses three.js under the hood (see https://threejs.org/docs/).  In that ecosystem, there are a huge number of very good special-purpose geometries and objects and materials defined beyond anything we'd ever introduce as a shape description in SceneGraph. So far, we've solved that by adding special-purpose `SetObject` overloads (or pseudo-overloads) -- e.g. `Meshcat::SetLine` to add a linestrip.

Functions like that are fine, but it means that anytime a user wants to access a not-yet-wrapped feature of three.js objects, they/we need to add a new Meshcat C++ function and C++ msgpack definition and bind it into pydrake and ... etc.

As an alternative, we should offer an "escape hatch" for advanced users who want access to everything three.js has to offer, where they just hand us the JSON object they want and we pass it along, unchanged.

To be clear -- we would still aim to cover the common use cases (e.g., linestrips) with sugared-up functions that make the simple things easy.  The goal here is to make the advance things at least _possible_.

**Describe the solution you'd like**

Something like:

```c++
 /** Sets the 3D object at a given `path` in the scene tree to the given shape,
  per the https://threejs.org/docs/#api/en/loaders/ObjectLoader threejs format. */
  void SetObjectRaw(std::string_view path, std::string_view threejs_object_json);
```

**Describe alternatives you've considered**

The only one I can think of is the existing condition: add bespoke functions one by one, for each kind of Object3D a user wants.

SeanCurtis-TRI, 2024-03-11:
Just to pin a relevant thought here:

One of the things sugar does is handle the difference between Drake's z-up world and three.js's y-up world. When we add this feature, we'll want to make sure that responsibility abundantly clear in its documentation.

siddancha, 2025-02-17:
> **Describe alternatives you've considered**
> The only one I can think of is the existing condition: add bespoke functions one by one, for each kind of Object3D a user wants.

Another alternative could be to expose the structs in [meshcat_types_internal.h](https://github.com/RobotLocomotion/drake/blob/master/geometry/meshcat_types_internal.h), such as `BoxGeometryData`,  `MaterialData` to the user. This would be pretty useful since specifying a custom SetObject command with C++ type-checks is easier than writing a JSON string where it might be harder to format the string, and catch bugs at compile time. The only issue is that the types live in an "internal" namespace. I'd be super happy if I could re-use the existing type machinery to, say, implement SetObject for a PCL Pointcloud from outside Drake.

jwnimmer-tri, 2025-02-17:
That serialization is an implementation detail that is unreasonable to try to stabilize.  Count on it remaining internal-use-only.  Users who want type checking for their bespoke geometry can copy / re-implement / customize to their needs.

This particular issue is about bespoke geometry, so the most-generic solution is what we want here.  If there is frequent use of everyday threejs types, we can add more first-party SetObject wrapper calls to support that (under a different issue).

We _could_ probably expose something like Meshcat::GenerateUuid() if that's helpful for users who are adding bespoke items.

> JSON string where it is harder to format the string, ...

Don't format the string by hand.  Use a json serialization library such as nlohmann json.

nepfaff, 2025-02-25:
This would be amazing for customizing the lighting. I would like to play with the different light options that tree.js offers.