RobotLocomotion/drake #20943: Provide MbP configuration option to use RpyFloatingJoints rather than QuaternionFloatingJoints

sherm1, 2024-02-14:
PR #20912 provides a roll-pitch-yaw floating joint as an alternative to the existing quaternion floating joint. Currently MultibodyPlant implicitly adds QuaternionFloatingJoints to mobilize free bodies. MbP should have an option that would make it add RpyFloatingJoints instead.

Alejandro's branch behind abandoned PR #19065 has a possible implementation using `MultibodyPlant::set_default_floating_joint_type()` with the corresponding `MultibodyPlantConfig` option.

N.B. Most of the free body code in MbP assumes that the bodies' inboard joint type is QuaternionFloatingJoint and will break if another joint type is present. That code needs to be generalized to use any suitable floating joint.

cohnt, 2024-09-10:
Just dropping by to say that this feature would be super useful for some of the stuff we're doing with IrisInConfigurationSpace when there are floating bodies. Do you have a sense of how time-consuming this would be to implement?

jwnimmer-tri, 2024-09-10:
BTW I just noticed one thing in the overview:

> N.B. Most of the free body code in MbP assumes that the bodies' inboard joint type is QuaternionFloatingJoint and will break if another joint type is present. That code needs to be generalized to use any suitable floating joint.

This is true, but also note that per https://github.com/RobotLocomotion/drake/issues/19269#issuecomment-1521098480 nearly all of those functions should just be deprecated for removal.  I suppose we don't want to wait for that deprecation timeline to expire in order to start working on this issue, but it does raise the option that those methods could _throw_ in case the plant setting is changed to "rpy" instead of "quat", to avoid investing the time in fixing up doomed code.

sherm1, 2024-09-10:
I'm working on a PR that will fix MbP internals so that floating bodies can have rpy joints. (The new topology code has global and per-model instance options to use rpys rather than quaternions for floating joints.) After the internals are working we can figure out how to convey that option in practical cases.

sherm1, 2024-11-11:
The C++ API for this is now in Drake master: [SetBaseBodyJointType()](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#a7c9672b405425259b192131304f14f9c).

Python, MbP config, and parsing access remain to be done.