RobotLocomotion/drake #20945: Integrator System Does Not Check Context Validity

cohnt, 2024-02-14:
### What happened?

When calling `integrator::set_integral_value`, the method computes the expected size of the state vector by querying the given context, but it does not check that the context is for the correct system. So if you pass in a diagram context, instead of throwing an error for passing in the wrong context, it throws an error for being given a vector of incorrect size (even if the size of the vector matches the size of the integrator's internal state). See [this line](https://github.com/RobotLocomotion/drake/blob/ddb32f9dba9db9984b9292ad5227a8911a5950e7/systems/primitives/integrator.cc#L25) of the definition for details.

jwnimmer-tri, 2024-02-15:
Specifically, the missing call is `this->ValidateContext(context);`.

cohnt, 2024-02-15:
I'm happy to go through all the primitives and make sure they all call `this->ValidateContext(context)` if you would like, since it looks like at least a few others have that problem. If the code change is that simple, I should be able to do it myself.

jwnimmer-tri, 2024-02-15:
Sure, tou're welcome to open a PR if you like, and thanks!

Note that callbacks coming from the system framework into a leaf system are already guaranteed to have been checked.  So protected and private functions (e.g., `Integrator::DoCalcVectorOutput`) don't need that new line -- only public functions need it.

PR #20958 (2024-02-17): Primitives context checks
Certain public methods of various primitive systems use a context without checking its validity. This can lead to downstream errors, such as the one encountered in #20945. I've gone through all the primitive systems and added context validation to the cases where it's not being checked, but values from the context are being used.

Closes: #20945.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20958)
closed 2024-02-17 by jwnimmer-tri