RobotLocomotion/drake #20948: Allow AddModel::default_free_body_pose with a non-world frame

jwnimmer-tri, 2024-02-14:
**Is your feature request related to a problem? Please describe.**

During scene authoring, we frequently want to set a default configuration for `q` as part of defining the scene.  To achieve that, we currently offer two optional fields of the `AddModel` directive: `default_joint_positions` and `default_free_body_pose`.

The `default_joint_positions` sets the configuration of the joints _inside_ of the model that we just added.  (This part seems to be working fine, and it not really relevant to this ticket.)

The `default_free_body_pose` sets the configuration of the _free bodies_ of the model that we just added.  (In many common cases, there is only one body.)  Specifically, it calls `MbP::SetDefaultFreeBodyPose` on the given bodies, which is documented as follows:

```c++
  /// Sets the default pose of `body`. If `body.is_floating()` is true, this
  /// will affect subsequent calls to SetDefaultState(); otherwise, the only
  /// effect of the call is that the value will be echoed back in
  /// GetDefaultFreeBodyPose().
  /// @param[in] body
  ///   RigidBody whose default pose will be set.
  /// @param[in] X_WB
  ///   Default pose of the body.
  void SetDefaultFreeBodyPose(const RigidBody<T>& body,
                              const math::RigidTransform<double>& X_WB);
```

In other words, assuming that nobody comes by later and adds a joint or weld to that body, then the given default pose will be used.

This is _sufficient_ to set the free body poses, but is not convenient.  Because the MbP uses `X_WB`, the base frame of the pose _must_ be the world, so we have this limitation on `AddModel`:

```c++
    for (const auto& [body_name, pose] : default_free_body_pose) {
      if (pose.base_frame) {
        drake::log()->error(
            "add_model: `default_free_body_pose` must not specify a "
            "`base_frame`; the pose is always in the world frame.");
        return false;
      }
```

In many cases, the most natural phrasing on the default pose will be w.r.t. some other frame in the scene (e.g., the center of a table, or the middle of the sink, or a clutter bin off to the side, etc.).  The user _can_ in theory compute the sequence of rigid transforms to get the value they need, but often ends up being very inconvenient.

**Describe the solution you'd like**

Remove the cited limitation in `AddModel`, and therefore all the `base_frame` to be specified in an `AddModel` directive.

**Describe alternatives you've considered**

Add some entirely new syntax?  Not sure.

jwnimmer-tri, 2024-02-14:
It will require a little bit of exploration how to deal with a pre-Finalize MbP in this case.

My initial thought is that a non-world base frame should cause the DMD parser to immediately add a `QuaternionFloatingJoint` from the base_frame to the body.  (\CC @sherm1 FYI)

Thinking aloud as I write this up, I also wonder if another missing use case is to use something other than the body frame for body end of the new joint.  In other words, the user might want to specify frames A and B by name (where B is rigidly affixed to the newly-added body) and then set the default pose X_AB.  I guess we could do that by backing out X_BBo from their default pose, though.  We'd still need somewhere for the user to tell us the body frame name, though.

jwnimmer-tri, 2024-02-14:
> We'd still need somewhere for the user to tell us the body frame name, though.

I suppose we could allow the map key to be interpreted as either a frame name or a body name, as in:

```yaml
- add_model:
    name: simple_model
    file: package://process_model_directives_test/simple_model.sdf
    default_free_body_pose:
      frame:
        base_frame: table::center
        translation: [1, 2, 3]
        rotation: !Rpy { deg: [5, 6, 7] }
```

jwnimmer-tri, 2024-03-04:
> Thinking aloud as I write this up, I also wonder if another missing use case is to use something other than the body frame for body end of the new joint.

Yes, this is actually a real requirement.

The name of the map key `default_free_body_pose` should allow for either body names or offset frame names.  (If that leads to some kind of implementation challenges, we'll need to circle back and regroup with some new syntax proposal.  One way or another, we do need to be able to specify an offset frame on the child end of the floating joint.)

rpoyner-tri, 2024-03-19:
To clarify, this is only a feature for DMD.yaml files? No other model formats need support this?

jwnimmer-tri, 2024-03-19:
Yes!

At some point the #17223 will need to learn how to convert this, but that's not part of this ticket.

SeanCurtis-TRI, 2024-03-28:
As of 3/28/2024, I'm working on this.

jwnimmer-tri, 2024-04-03:
The comments in #21245 make me a little bit worried that this is more complex than I imagined.  I expected that this would be as simple as adding a new floating joint during dmd parsing?  I don't yet see where CreateDefaultContext comes in to play.  You're not adding stochastic Expression support, are you?  That's not in scope here.  Possibly we should sync up about what the design looks like.

sherm1, 2024-04-03:
This is a trickier project than it first appears. It is a special case of the more general problem of solving for initial conditions to satisfy constraints. Here the constraint is that we want to place a free body F in a pose defined by some body B other than World. But ... in general we can't know how B will be located in space until we have values for the coordinates q that affect B's pose. And B could be another free body or dependent on some other free body's pose, introducing coupling among the free body poses. And actually we want to express the desired pose in terms of arbitrary offset frames fixed to F and B. Offset frames are parameterized, so we can't know where they are until we have values for parameters.

Assuming we forbid cycles in the free body pose dependency graph as Sean proposes, this can be solved iteratively by setting parameters to fix the offset frame poses on their bodies, topologically sorting the dependencies, then iteratively placing the least-dependent free body, re-evaluating kinematics, place the next free body, etc. Currently that has to be done under SetDefaultState() which doesn't have write access to the Context, making the implementation awkward (but doable).

A more general alternative would be to leave CreateDefaultContext() alone and add a follow-up phase like SolveForInitialConditions() that would have write access to the context and be able to satisfy other constraints, deal with error conditions, handle conflicts, etc.

SeanCurtis-TRI, 2024-04-03:
Summary of f2f:

Jeremy was solving a different problem than Sean.

Sean: for a free body (moved by a floating joint between the body and *any* other body), allow its *initial* pose by describing it w.r.t. to an arbitrary frame. This introduces the need to carefully evaluate kinematics iteratively in a proper order so that every pose is properly defined (topological sort on a graph). 

Jeremy: someone has a free body that they want to pose w.r.t. another object (aka frame). Typically, they specify *no* joint and this leads to the plant introducing a floating joint between that body and the world (which is why the default pose is measured and expressed in the world). Instead, at the time someone explicitly requests to pose a free body w.r.t. a particular frame, a floating joint is introduced between the body and that frame. Then, by construction, the pose will be measured and expressed w.r.t. that reference frame. This is a much quicker path to achieve this more limited end.

A special note, dynamically speaking, there's really no difference between building a floating joint between free body and the world frame or any other frame; forces don't get transmitted across floating joints. However, kinematically, there is a difference. Changing any of the qs that lie upstream of the floating joint will change the world position of the free body. This may have an impact on planning. To kinematically move the arm and leave the body stationary, the qs of the floating joint would have to be perpetually updated to offset movement in the upstream kinematic chain.

For now, we'll pursue the "put in a floating joint" strategy as sufficient for the most immediate needs, but anticipate that the more general problem will eventually need to be solved.

closed 2024-04-09 by SeanCurtis-TRI