RobotLocomotion/drake #20953: Switch Ubuntu wheel builds to Jammy

jwnimmer-tri, 2024-02-15:
Here in the list of targets ...

https://github.com/RobotLocomotion/drake/blob/295eb0f07591f932e06ffd5c3576b51409ebfd7d/tools/wheel/wheel_builder/linux.py#L30-L34

... we drop 3.8 and 3.9 from the list entirely, and for 3.10 and up we change to build the wheel on jammy, and test it on jammy (except 3.12, which I guess we'll still test on Mantric for now.

Note that this will also probably involve changes to `download_release_candidate`, the release playbook, and drake-ci.

mwoehlke-kitware, 2024-02-15:
Oh, good! A long while back I'd filed https://github.com/pypa/auditwheel/issues/388, but allegedly it got fixed a little over a year ago already (and I guess we missed it, or possibly chose to stay on Focal for other reasons). So, hopefully switching to Jammy as the build host should be relatively painless.

jwnimmer-tri, 2024-02-15:
> ... possibly chose to stay on Focal for other reasons ...

Yes, we previously needed to keep Focal as our docker base to enable all `*.whl` files to be installed on our users' Focal systems.

As of today, it's OK to upgrade our minimum supported OS to Jammy.

PR #20961 (2024-02-18): Move wheel creation to Jammy, drop older Python
Change Linux wheel builds to use Jammy, rather than Focal, as the build host. Remove Python targets that are older than Jammy. Remove Focal-specific hacks from wheel builds.

Fixes #20953.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20961)
closed 2024-02-18 by jwnimmer-tri