RobotLocomotion/drake #20954: Turn off Focal CI

jwnimmer-tri, 2024-02-15:
We definitely want to turn it off (as a precondition for ripping out the support code inside Drake).

However, I suppose before we do that we should make sure that:

(1) We aren't losing any of the config matrix (i.e., everything is twinned on Jammy already).  Can you check into this for me?

(2) Maybe keep a record of how we had split up Focal/Jammy, so that when we do Jammy/Noble CI spin up next month we know what jobs we'll want.

jwnimmer-tri, 2024-02-15:
FYI https://github.com/RobotLocomotion/drake/issues/20953 re: wheel builds.

jwnimmer-tri, 2024-02-18:
I did the cross-checking for "Unprovisioned" builds:

> (1) We aren't losing any of the config matrix (i.e., everything is twinned on Jammy already). Can you check into this for me?

I checked the list of [experimental unprovisioned](https://github.com/RobotLocomotion/drake/blob/jenkins-jobs-experimental/request-jobs-unprovisioned.txt) and Jammy is already on par with Focal in terms of which job flavors are defined -- except for the wheel build (#20953).  For the wheel, the CI image is OK to stay on Focal until https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/113 is fixed.  In the meantime, we can still land #20961 so that that created wheels are >= Jammy even though the CI base is Focal (to launch Docker).

I also checked the trigger schedule for unprovisioned, and Jammy is already on par with Focal, except for:
(a) Jammy has "mirror" and "documentation" builds and Focal does not -- this is good, no changes needed; when we add Noble, we'll still keep this two as Jammy-only.
(b) Focal has the Wheel build and Jammy does not -- this is not an immediate problem, per the above;
(c) The `linux-focal-unprovisioned-gcc-bazel-continuous-packaging` is Focal-only (the Jammy flavor is Nightly).  **We need to switch Continuous Packaging to use Jammy, not Focal**.  When we add Noble, we do not need to trigger Noble as Continuous; it's OK to just have 1 flavor be Continuous.

jwnimmer-tri, 2024-02-18:
In anticipation of the changes to pre-merge triggering, I've edited Drake's GitHub settings for branch protection rules so that these builds are no longer required to pass:
- `linux-focal-clang-bazel-experimental-release`
- `linux-focal-gcc-bazel-experimental-everything-release`
- `linux-focal-gcc-cmake-experimental-release`

jwnimmer-tri, 2024-02-18:
For the new pre-merge triggers, let's do this:
- `linux-jammy-clang-bazel-experimental-everything-release` (unchanged)
- `linux-jammy-clang-bazel-experimental-leak-sanitizer` (unchanged)
- `linux-jammy-gcc-bazel-experimental-debug` (unchanged)
- `linux-jammy-gcc-bazel-experimental-everything-release` (**switched over from focal**)
- `linux-jammy-gcc-cmake-experimental-release` (**switched over from focal**)
- `linux-noble-clang-bazel-experimental-release` (switched over from focal)
- `linux-noble-gcc-bazel-experimental-release` (switched over from jammy)

Of course the Noble jobs will not be pre-merge triggered until they have been defined and are passing, but those will be the aim.  In the meantime, we'll only use the other five ubuntu jobs as pre-merge.

This will also impact what we run in Continuous Production.  (Anything pre-merge should also be in Continuous, not Nightly.)

svenevs, 2024-02-21:
> I also checked the trigger schedule for unprovisioned, and Jammy is already on
> par with Focal, except for:
>
> (a) Jammy has "mirror" and "documentation" builds and Focal does not -- this
>     is good, [...]

For (b), I went ahead and made #21007 to track that away from this ticket.
For (c), [completed just now](https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/114),
first build is [currently running](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-unprovisioned-gcc-bazel-continuous-packaging/1/consoleFull).

jwnimmer-tri, 2024-02-21:
Sounds great.

And just to be clear (hopefully it was obvious) -- for Production builds, I didn't do very much digging.  I posted the recipe for pre-merge triggering up above, but in terms of the permutation matrix and triggering, I don't have any specific advice.

svenevs, 2024-02-21:
> For the new pre-merge triggers, let's do this:
>
> - linux-jammy-clang-bazel-experimental-everything-release (unchanged)
> - linux-jammy-clang-bazel-experimental-leak-sanitizer (unchanged)
> - [...]

Ok, there was a little bit of confusion about the `exclude` logic.  The focal
pre-merge jobs have been removed and the available jammy replacements created.
See https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/115

> - **linux-jammy-gcc-bazel-experimental-everything-release (switched over from focal)**
> - **linux-jammy-gcc-cmake-experimental-release (switched over from focal)**

These two were added in the jammy category, and the PR flavor of the focal
version removed.

> - **linux-noble-clang-bazel-experimental-release (switched over from focal)**

Focal version dropped, jammy promoted to PR.

> - linux-noble-gcc-bazel-experimental-release (switched over from jammy)

TBD.  I've also created an issue for us to track the "noble pre-merge" scenario
specifically so that you do not need to recreate the list here in #21008.

@jwnimmer-tri you should be able to update the required checks on the GitHub
side now.  Please let us know if something unexpected has occurred so we can
fix it ASAP!

svenevs, 2024-02-21:
> This will also impact what we run in Continuous Production. (Anything pre-merge should also be in Continuous, not Nightly.)

FYI I'm fixing this right now, #21010 seems like it has the right 9 jobs we want.

jwnimmer-tri, 2024-02-21:
I updated the GitHub pull request requirement checks.

Looking at https://github.com/RobotLocomotion/drake/pull/21010 it seems like I did it correctly, but please cross-check me.

svenevs, 2024-02-21:
Continuous jammy is up and running, I launched the new jobs just now so the triggers should match up.  https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/116 

> Looking at https://github.com/RobotLocomotion/drake/pull/21010 it seems like I did it correctly, but please cross-check me.

Looks right to me!

-------------------

@BetsyMcPhail is going to drop the other focal jobs (except for wheel) today IIUC.

BetsyMcPhail, 2024-02-22:
Nightly unprovisioned jobs are duplicated for Focal and Jammy with the following exceptions:

- `wheel` Focal only
- `documentation` Jammy only
- `linux-jammy-unprovisioned-gcc-bazel-nightly-release` Jammy only (??)

BetsyMcPhail, 2024-02-22:
Current state of Provisioned Nightly jobs:

- linux-jammy-clang-bazel-nightly-address-sanitizer 
- linux-jammy-clang-bazel-nightly-coverage 
- linux-jammy-clang-bazel-nightly-everything-leak-sanitizer 
- linux-jammy-clang-bazel-nightly-everything-thread-sanitizer 
- linux-jammy-clang-bazel-nightly-everything-undefined-behavior-sanitizer 
- linux-jammy-clang-bazel-nightly-everything-valgrind-memcheck 
- linux-jammy-clang-bazel-nightly-leak-sanitizer 
- linux-jammy-clang-bazel-nightly-thread-sanitizer 
- linux-jammy-clang-bazel-nightly-undefined-behavior-sanitizer 
- linux-jammy-clang-bazel-nightly-valgrind-memcheck 
- linux-jammy-gcc-bazel-nightly-coverage 
- linux-jammy-nightly-cache-server-health-check 

- linux-focal-clang-bazel-nightly-everything-debug 
- linux-focal-clang-bazel-nightly-everything-release 
- linux-focal-clang-bazel-nightly-gurobi-release 
- linux-focal-clang-bazel-nightly-mosek-release 
- linux-focal-clang-bazel-nightly-snopt-release 
- linux-focal-gcc-bazel-nightly-everything-debug 
- linux-focal-gcc-cmake-nightly-debug 
- linux-focal-gcc-cmake-nightly-everything-debug 
- linux-focal-gcc-cmake-nightly-everything-release 

Interestingly, there is no overlap here. All focal-only jobs have a jammy-experimental equivalent. For now, moving all nightly production to Jammy.

BetsyMcPhail, 2024-02-22:
All that is remaining for focal is experimental. We already did a complete pass through pre-merge, continuous and nightly, so these can all go with no additional checks (with the exception of the wheel build!)

BetsyMcPhail, 2024-02-22:
All focal jobs with the exception of wheel builds have been removed!

closed 2024-02-22 by BetsyMcPhail