RobotLocomotion/drake #20957: Multiple `CameraConfig`s with reused renderers do not `show_rgb` correctly

dmcconachie-tri, 2024-02-15:
### What happened?

When instantiating multiple cameras with the same render engine, enabling `show_rgb` generates "flickering" behavior from the display window. This comes in 2 variants. If both cameras have `show_rgb` enabled, the window flickers back and forth between the 2 (or more) views. If only one camera has `show_rgb` enabled, the window flickers open and closed.

To reproduce apply this diff
```
diff --git a/examples/hardware_sim/example_scenarios.yaml b/examples/hardware_sim/example_scenarios.yaml
index a033ed6847..29cc2b6843 100644
--- a/examples/hardware_sim/example_scenarios.yaml
+++ b/examples/hardware_sim/example_scenarios.yaml
@@ -52,9 +52,16 @@ Demo:
   cameras:
     oracular_view:
       name: camera_0
+      show_rgb: true
       X_PB:
         translation: [1.5, 0.8, 1.25]
         rotation: !Rpy { deg: [-120, 5, 125] }
+    secondary_view:
+      name: camera_1
+      show_rgb: true
+      X_PB:
+        translation: [1.5, 0.8, 0.25]
+        rotation: !Rpy { deg: [-120, 5, 125] }
   model_drivers:
     iiwa: !IiwaDriver
       hand_model_name: wsg
```
and run `bazel run //examples/hardware_sim:demo_cc`

### Version

295eb0f07591f932e06ffd5c3576b51409ebfd7d

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

jwnimmer-tri, 2024-02-15:
Thanks for writing this up.  IIRC it's been discussed in slack previously, but never elevated here.

Let me double-check your use case, though, so we can aim this the right way ...

From one point of view, the general idea behind "show_rgb" is "Help me debug GPU problems as I'm hacking on RenderEngine code".  It's an engine-specific outcome.  Some engines don't even support it.  In other engines the image is upside down, etc.  Label images also right with rgb images in the same flickering way mentioned upthread.

If your use case is "As a simulation author, I want to watch what my simulated cameras are seeing" then under the above paragraph the answer here isn't "remove show_rgb flicker" but rather "instrument some of the Image ports inside my Diagram for live viewing".

For that use case, the nominal answer today is to publish the images over LCM, and use an LCM image viewer to monitor them.  Anzu already has a performant LCM image viewer you could use.  We also have #18862 underway to show the images in Meldis (a web browser).

dmcconachie-tri, 2024-02-15:
My use case is the second; "As a simulation author I want to add cameras. I want to verify what the cameras can see". A pointer to the anzu viewer would be useful as I am not finding it.

jwnimmer-tri, 2024-02-15:
+@zachfang per https://drake.mit.edu/issues.html leads.

zachfang, 2024-02-15:
I had a similar issue before and will look into that and propose/submit possible solution.

zachfang, 2024-02-20:
I have run through different configurations (3 render engines & `label: {true, false}`), and, as Jeremy stated, the behavior is RenderEngine-dependent. However, the flickering was consistent when multiple cameras sharing the same renderer.

Since the flag is designed for debug usage while a proper image viewer is proposed, I think the action item is to add documentation to capture the discussion here to avoid future confusion. I can make a PR this week.

PR #21018 (2024-02-26): Update show_rgb documentation
Close #20957

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21018)
closed 2024-02-26 by sherm1