RobotLocomotion/drake #20962: Possibly spurious "changing the illustration roles in the context" warning

jwnimmer-tri, 2024-02-17:
### What happened?

There is a warning in our code:

https://github.com/RobotLocomotion/drake/blob/fefe2570d2f591b142aa4420d1f4f54d072c704e/geometry/scene_graph.cc#L384-L389

However, that issue is closed and the description in the issue reads to me like the problem would be truly gone by now.

If the problem is gone, then we should remove the warning (in both the c file and the header file).

If the problem still persists, we should either fix it, or at minimum clarify the comments so that it's clear why it's still relevant now that `//tools:drake_visualizer` has been nuked from orbit.

This is follow-up from #13597 that got left behind accidentally.

### Version

v1.26.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
I didn't try to collect any in-site logs yet.
```

SeanCurtis-TRI, 2024-02-19:
Note: resolution also includes a comment on the function's declaration in the header file. Don't forget to do both.

sadraddini, 2025-12-01:
> Nope. Neither the console spam nor doxygen warning should be relevant.

Remove both?