RobotLocomotion/drake #20978: Move `PairwiseIntersectionsContinuousJoints` to public API

sadraddini, 2024-02-19:
**Is your feature request related to a problem? Please describe.**
The function `PairwiseIntersectionsContinuousJoints` is used to detect intersections and the required offsets between convex sets based on the indices of continuous revolute joints. It means it can move the convex set in those dimensions by multiples of 2*pi to make them intersect. 

This function is not in the public API.

The use cases for this from a public user:
1- When using `Subgraph::AddRegions` as in [here](https://drake.mit.edu/doxygen_cxx/classdrake_1_1planning_1_1trajectory__optimization_1_1_gcs_trajectory_optimization.html#aff9544b6375a592b1b9bda5eb1149a6f), there is an argument to give edge offsets. But there is no public API for computing those. 

2- This forces the user the other overload, which, under the hood, computes all intersections and offsets from scratch.  The user may simply know most of the intersections and edges and does not want to waste time on re-computing those. This comes handy when one has to quickly set up a modified GCS without re-computing all edges, etc. 

3- When creating Iris regions, I always check the connectivity. It is useful to just check the connection between a newly added region and previous regions in the case that some joints are continuous revolute. 

**Describe the solution you'd like**
Move this to the public API.

**Describe alternatives you've considered**
Messy workaround: The user sets up a GCS, registers new regions for which intersections are unknown as new subgraphs, ask to add edges, and check the existence of edges. 

@cohnt let me know what you think.

wrangelvid, 2024-02-20:
@RussTedrake I would say @cohnt is better suited to tackle this since he introduced continuous joints.

cohnt, 2024-02-21:
I've moved these functions to geodesic convexity and made them public in #21011. But now that they're not internal, I need to add a bunch of test cases, which I might not be able to do for a little while.

In the meantime, you can use that branch (and if you want to add the test cases yourself, Sadra, then feel free).

sadraddini, 2024-02-24:
Alright, I will be adding the necessary unit tests today.

PR #21011 (2024-02-28): Make PairwiseIntersectionsContinuousJoints Non-Internal
I've also moved these to the geodesic convexity file, to avoid cluttering up GcsTrajectoryOptimization's interface.

Still need to add test cases, now that they're public functions. Once I've done so, I'll call for review.

Resolves #20978.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21011)
closed 2024-02-28 by sadraddini