RobotLocomotion/drake #20982: Allow `TrajectorySource` with `output_derivative_order` > 0 without requiring `MakeDerivative`

nepfaff, 2024-02-19:
**Is your feature request related to a problem? Please describe.**
The current implementation of `TrajectorySource` requires the trajectory to implement `MakeDerivative` (see [here](https://github.com/RobotLocomotion/drake/blob/master/systems/primitives/trajectory_source.cc#L30)). However, there are cases when it is more natural/ easier to provide `DoEvalDerivative` only. An example is `PathParameterizedTrajectory`. I had a similar situation with one of my custom trajectories, where providing `DoMakeDerivative` added unnecessary overhead while `DoEvalDerivative` was `O(1)` anyways.

**Describe the solution you'd like**
Allow `TrajectorySource` with `output_derivative_order` > 0 without requiring `MakeDerivative`. In particular, it should support `PathParameterizedTrajectory`.

RussTedrake, 2024-02-20:
It seems like the first step towards this would be to split `Trajectory::has_derivative` to enable declaring support for `EvalDerivative` independently from `MakeDerivative`.

jwnimmer-tri, 2024-02-20:
> It seems like the first step towards this would be to split Trajectory::has_derivative to enable declaring support for EvalDerivative independently from MakeDerivative.

I would rather say that any class that implements one of the two should always implement both.  In other words, `PathParameterizedTrajectory` should implement `MakeDerivative`.  We could even have a common helper class `DerivativeTrajectory` that wraps an underlying `Trajectory` with an `int order`, calling `underlying.EvalDerivative` for the actual logic.

jwnimmer-tri, 2024-02-20:
In fact, `PathParameterizedTrajectory` is outright wrong.  The docs for `Trajectory::has_derivative()` say that if you return true, you must provide an implementation for _both_ functions, yet it fails to do so.

RussTedrake, 2024-02-20:
I agree that PathParameterizedTrajectory is wrong in this way.  
We also have a request for a FunctionHandleTrajectory in #20699.  Implementing that could make the DerivativeTrajectory class even easier.

PR #21036 (2024-02-25): Implements DerivativeTrajectory.
Resolves #20982.

+@nepfaff for feature review, please.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21036)
closed 2024-02-25 by jwnimmer-tri