RobotLocomotion/drake #20988: Appearance of proximity meshes in meshcat

SeanCurtis-TRI, 2024-02-20:
There are a number of artifacts vis a vis how proximity geometry gets visualized. This issue tracks a number of related issues.

These are considered obvious improvements/requirements:
- [ ] #20618
- [ ] #20983
- [ ] #20984

These ideas are more speculative, with speculative benefits:
- [ ] #20986
- [ ] #20987

jwnimmer-tri, 2024-11-02:
@SeanCurtis-TRI my preference would be to prioritize all of those issues independently, and just close this Epic?  At least for me, it's not adding value.  If you like it we can keep it though.

SeanCurtis-TRI, 2024-11-04:
Feel free to close it. The issues themselves maintain a record of this. The ability to get from one of those issues to other related issues is now baked in. So, whatever value this issue serves, it serves it just as well closed.

closed 2024-11-04 by SeanCurtis-TRI