RobotLocomotion/drake #20989: Add binding for `SingleOutputVectorSource`

nepfaff, 2024-02-20:
**Is your feature request related to a problem? Please describe.**
I want to subclass [SingleOutputVectorSource](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_single_output_vector_source.html) in pydrake.

**Describe the solution you'd like**
A binding for `SingleOutputVectorSource`.

**Describe alternatives you've considered**
I tried opening a PR for the binding, but I failed. My attempt added the following just below [here](https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/systems/framework_py_systems.cc#L297). I wanted to open this issue to ask whether this is desired before spending too much time getting the binding right.
```python
 auto single_output_vector_source_cls =
        DefineTemplateClassWithDefault<SingleOutputVectorSource<T>, LeafSystem<T>>(
            m, "SingleOutputVectorSource", GetPyParam<T>(),
            doc.SingleOutputVectorSource.doc);
    single_output_vector_source_cls  // BR
        .def(py::init<int>(), py::arg("size"),
            doc.SingleOutputVectorSource.ctor.doc_1args_size)
        .def(py::init<const BasicVector<T>&>(), py::arg("model_vector"),
            doc.SingleOutputVectorSource.ctor.doc_1args_model_vector);
        .def("get_output_port", &SingleOutputVectorSource<T>::get_output_port,
            py_rvp::reference_internal, doc.SingleOutputVectorSource.get_output_port.doc)
        .def("DoCalcVectorOutput", &SingleOutputVectorSource<T>::DoCalcVectorOutput,
            py::arg("context"), py::arg("output"),
            doc.SingleOutputVectorSource.DoCalcVectorOutput.doc);
```

For now, I can inherit from `LeafSystem` directly.

RussTedrake, 2024-02-20:
I think the recommendation is to stick with deriving from `LeafSystem`.  In order to do this properly, you would need to introduce a "trampoline" class, like PyLeafSystem.  `VectorSystem` and `SingleOutputVectorSource` were more important earlier, when writing a `LeafSystem` for these cases was quite a burden.  Now one can write the `LeafSystem` directly with only a few lines.

Is there a reason you want to derive from this class?

nepfaff, 2024-02-20:
That is good to know. Thanks! In that case, it seems more effort than it's worth. Should we close this one?

Deriving from this seemed cleaner than from LeafSystem based on the C++ documentation of that class.

jwnimmer-tri, 2024-02-20:
> Should we close this one?

If nothing else, we should clarify its documentation to steer others away from the same trap you fell into.  We should probably even mark it as "To be deprecated", or at least "Not useful anymore, don't bother trying to use it", etc.

RussTedrake, 2024-02-20:
+@jwnimmer-tri to deprecate / demarcate (or delegate).

jwnimmer-tri, 2024-02-20:
Roger wilco.  I've updated the title accordingly.

jwnimmer-tri, 2024-03-16:
Along similar lines, we should update the docs for VectorSystem to point people away from it, since it's not that relevant anymore.

jwnimmer-tri, 2026-09-03:
For the record, the next steps here are to rework `primitives/constant_vector_source.h` and `primitives/trajectory_source.h` to use `LeafSystem` directly (instead of `SingleOutputVectorSource`), and deprecate `SingleOutputVectorSource`.

closed 2026-09-08 by SeanCurtis-TRI