RobotLocomotion/drake #21003: Option to add slack variables on GCS edges

bernhardpg, 2024-02-21:
**Is your feature request related to a problem? Please describe.**
I would like a more flexible way to formulate costs and constraints on the edges in GCS using slack variables. Right now, the slack variables need to be defined as part of the convex set dimension.

For instance, say one would like to implement the cost:
```
min max(1/x_1, 1/x_2) st. x_1, x_2 >= 0
```
over a two-dimensional convex set (x_1 and x_2).

This one can do with by adding the slack variable `s` and reformulating as follows:
```
min s st. s >= 1 / x_1, s >= 1 / x_2, x_1, x_2, s >= 0
```
which is readily encoded using RotatedLorentzConeConstraints:

```
min s st. s * x_1 >= 1, s * x_2 >= 1, x_1, x_2, s >= 0
```

The way to do this right now in GCS is the following:
1. Have some convex set in mind with ambient dimension 2
2. Extend the dimension of the set to 2 + 1 (to accommodate for the slack variable)
3. Add the RSOC constraint on the edges/vertices, using the convex set variables.

This approach has some problems:
- Imagine the scenario where one would like to implement i.e. a cost that is SOC representable on some of the edges incident to a vertex, but not all. This is not possible with the current workflow, as the slack variable is embedded in the convex set itself.
- The slack variables may not naturally be bounded.

**Describe the solution you'd like**
I think it would be helpful to have a function on the edges which adds and returns a slack variable directly to the underlying internal `MathematicalProgram` in GCS, called something like `Edge.CreateSlackVariable(...)`.

The `Edge.AddConstraint(binding)` and `Edge.AddCost(binding)` functions would then have to figure out whether the variables in the binding are "convex set" variables or slack variables, and accordingly retrieve the spatial flow variables (in the case of "convex set" variables) or just use the variable itself (in the case of slack variables).

I believe this should also be easy to extend to have a similar function for vertices.

bernhardpg, 2024-02-21:
@TobiaMarcucci @RussTedrake WDYT?

cohnt, 2024-02-22:
I think this feature would also allow the subspace mechanics in GcsTrajectoryOptimization to work with arbitrary convex sets, since adding point-in-set constraints for some convex sets requires adding additional variables.

RussTedrake, 2024-04-22:
See Remark 5.1 in Tobia's thesis.

RussTedrake, 2024-08-19:
I believe this is resolved by #21403.  @bernhardpg -- please re-open if I'm wrong.

closed 2024-08-19 by RussTedrake