RobotLocomotion/drake #21008: Noble pre-merge jobs

svenevs, 2024-02-21:
https://github.com/RobotLocomotion/drake/issues/20954#issuecomment-1951386653

The desired pre-merge jobs (when noble is operational):

- [X] linux-jammy-clang-bazel-experimental-everything-release (unchanged)
- [X] linux-jammy-clang-bazel-experimental-leak-sanitizer (unchanged)
- [X] linux-jammy-gcc-bazel-experimental-debug (unchanged)
- [X] linux-jammy-gcc-bazel-experimental-everything-release (switched over from focal)
- [X] linux-jammy-gcc-cmake-experimental-release (switched over from focal)
- [ ] linux-noble-clang-bazel-experimental-release (switched over from focal)
- [ ] linux-noble-gcc-bazel-experimental-release (switched over from jammy)

This ticket is for tracking the noble PR pre-merge jobs.

CC @jwnimmer-tri @BetsyMcPhail

jwnimmer-tri, 2024-03-16:
See also #20835.

jwnimmer-tri, 2024-05-23:
For clarity, I'm going to roll the remaining work here into #21475, so that all of the CI changes have an inventory in a single place.

closed 2024-05-23 by jwnimmer-tri