RobotLocomotion/drake #21017: get_iiwa_max_joint_velocities should return different values for iiwa7 and iiwa14

nepfaff, 2024-02-22:
### What happened?

This is my bad as I didn't notice the existence of [this function](https://github.com/RobotLocomotion/drake/blob/master/manipulation/kuka_iiwa/iiwa_constants.cc#L9C17-L9C46) when [updating the iiwa7 joint velocities](https://github.com/RobotLocomotion/drake/commit/e6384996ffc44f8d0866a5308c1da660f042237e).

The values can be found in the tables [here](https://github.com/RobotLocomotion/drake/tree/master/manipulation/models/iiwa_description#velocity-and-effort-limits-for-iiwa-7).

My suggested solution is to add a boolean argument to identify whether the iiwa is 7 or 14 and return different values accordingly. Any thoughts? 

### Version

d899c03

jwnimmer-tri, 2024-02-22:
What is the problem you're trying to solve?

On first look, that function appears to only be used by test programs.  Having it be hard-coded to iiwa14 is not a problem, in that light.

It seems to me like any code that's using `get_iiwa_max_joint_velocities`, instead of the limits loaded from the SDFormat file, is defective.

Maybe the bug is that function exists and tempts people to use it.  We should probably mark it "for testing only" and move it into the internal namespace.

jwnimmer-tri, 2024-03-19:
Closing as not planned.

closed 2024-03-19 by jwnimmer-tri