RobotLocomotion/drake #21019: `kIiwaLcmStatusPeriod` must be configurable

nepfaff, 2024-02-22:
### What happened?

[kIiwaLcmStatusPeriod](https://github.com/RobotLocomotion/drake/blob/master/manipulation/kuka_iiwa/iiwa_constants.cc#L27C14-L27C34) is currently a constant and is used by `BuildIiwaControl`, regardless of the iiwa control mode. However, the driver specifies different periods for different control modes. In particular, `[DrakeFRITorqueOnlyDriver](https://github.com/RobotLocomotion/drake-iiwa-driver/blob/master/kuka-driver/sunrise_1.14/DrakeFRITorqueOnlyDriver.java)` uses a period of `0.001` and all other drivers use a period of `0.005`. See [here](https://github.com/RobotLocomotion/drake-iiwa-driver/tree/master/kuka-driver/sunrise_1.14).

We should either select the period based on the `control_mode` or add it as an attribute to [IiwaDriver](https://drake.mit.edu/doxygen_cxx/structdrake_1_1manipulation_1_1kuka__iiwa_1_1_iiwa_driver.html). I'm leaning towards the later as it would make it easier for users to make low-level driver changes.

### Version

d899c03

sherm1, 2024-02-23:
@sammy-tri please reassign if you're not the right person to look at this

sammy-tri, 2024-02-27:
I think adding a new field to `IiwaDriver` (`status_period`?  `control_period`?) seems fine to me (and threading it through `BuildIiwaControl` and down into `SimIiwaDriver` seems like a reasonable plan to make this configurable.  I'm not sure I'll personally have time to do it in the near future though I would be willing to be feature reviewer on a PR.

jwnimmer-tri, 2024-11-01:
We are open to pull requests, but the Drake team is not planning to work on this.  We don't use IIWAs at all anymore.

closed 2024-11-01 by jwnimmer-tri