RobotLocomotion/drake #21023: visualization_install_tests vs macOS python3.12 upgrade

jwnimmer-tri, 2024-02-22:
The `visualization_install_tests` are somewhat brittle w.r.t. how we choose macOS python versions.

For now, I'm going to disable them on macOS to unbreak CI.

jwnimmer-tri, 2024-10-18:
@mwoehlke-kitware with the venv, I believe we can probably revert the #21024 now.  Could you give it a try?

CC @BetsyMcPhail

jwnimmer-tri, 2024-11-15:
It doesn't make any sense why Provisioned and Unprovisioned jobs would have a different result.  That needs some investigation.

BetsyMcPhail, 2024-11-15:
Provisioned (passed): https://drake-jenkins.csail.mit.edu/view/Pull%20Request/job/mac-arm-sonoma-clang-bazel-experimental-release/683/
Unprovisioned (failed): https://drake-jenkins.csail.mit.edu/view/Mac%20Sonoma%20Unprovisioned/job/mac-arm-sonoma-unprovisioned-clang-bazel-experimental-release/53/

nicolecheetham, 2024-11-15:
In both of the unprovisioned job attempts there was a failure with //:py/install_test. In addition they mention a failed result of 
pydrake/visualization/test/visualization_install_tests.py. However, in both provisioned jobs there is no mention of running pydrake/visualization/test/visualization_install_tests.py let alone fail.

jwnimmer-tri, 2025-02-24:
The fix might be to `execve` the venv-installed python interpreter (with `-m` foo etc.) instead of importing the module.