RobotLocomotion/drake #21025: Add port name prefix argument to `SimIiwaDriver`

nepfaff, 2024-02-22:
**Is your feature request related to a problem? Please describe.**
Currently, [SimIiwaDriver](https://github.com/RobotLocomotion/drake/blob/master/manipulation/kuka_iiwa/sim_iiwa_driver.cc#L62) exports iiwa ports with fixed names such as `position_commanded`. This makes it impossible to add multiple iiwas to the same builder with `SimIiwaDriver`. 

**Describe the solution you'd like**
I want the ability to prefix the port names. This is implemented in Russ' manipulation repo and works very well. See [here](https://github.com/RussTedrake/manipulation/blob/master/manipulation/station.py#L586) for prefixing `position_commanded` with `{model_instance_name}.`.
A simple solution would be to add a `port_prefix` argument with an empty string default value. I'd be happy to make this change if there are no counterarguments.

**Additional context**
I'm working on switching the logic in Russ' manipulation repo to use the Drake functions, such as `AddSimIiwaDriver`. The goal is for this to serve as RLG's common hardware setup.

jwnimmer-tri, 2024-02-22:
I don't understand.  The `SimIiwaDriver` port names are exactly as-intended, and do not cause the problem you describe.  You can put two instances of `SimIiwaDriver` into the same overall `Diagram`, and nothing will complain.

I anticipate that you have code somewhere that calls "export all of the ports that I can find".  _That_ code is where the name qualifiers are supposed to be added.

jwnimmer-tri, 2024-02-22:
No problem!  Glad we got it sorted quickly.