RobotLocomotion/drake #21030: [ci] arm64 support

jwnimmer-tri, 2024-02-23:
**Is your feature request related to a problem? Please describe.**

As #13514 gets ever closer, one question is how Drake would set up CI that covers it.

The help prepare, I'd like to get some ballpark of the ongoing additional AWS cost that would be required.

**Describe the solution you'd like**

The objective would be to add these jobs to CI:

```
linux-arm64-jammy-unprovisioned-gcc-bazel-nightly-everything-release
linux-arm64-jammy-unprovisioned-gcc-bazel-nightly-packaging
linux-arm64-jammy-unprovisioned-gcc-wheel-nightly-release
```

We would also define their `-experimental-` twin so we can check them in pull requests, but they would not be pre-merge (nor "Continuous").  Only "Nightly Production".

One caveat: for the first job ("everything") we might end up choosing to run only the subset of solvers that match "packaging" and "wheel", instead of all three solvers.  (So, e.g., `linux-arm64-jammy-unprovisioned-gcc-bazel-nightly-mosek-snopt-release`.) However, I don't imagine that tweak will affect the AWS bill very much.

We would **not** use the cache server for these jobs.  They are all built (& tested) from scratch every night.

Please put together a rough estimate of the new spending required, per month.  I imagine the easiest way would be to take a look at the build times for the current x86 workers to get an estimate of how many minutes each flavor would need (for a given number of cores), and then choose a similar graviton instance type and multiple by its spot cost.

It's fine to just give me the +Nightly cost.  I can add a fudge factor to estimate the additional PR builds that might be run.

**Describe alternatives you've considered**

If the unprovisioned cost (with no caching) comes back too high, I might consider adding provisioned images into the solution (with a weekly unprovisioned as backstop).  However, that requires more human time to babysit, so for now I'm keeping it off the table.

jwnimmer-tri, 2024-03-07:
Okay, thanks!

I suppose we don't need the "64" in the name, do we?  In macOS land we just say `mac-arm-...`.

So as my next request, please define some new experimental jobs:

```
linux-arm-jammy-unprovisioned-gcc-bazel-experimental-everything-release
linux-arm-jammy-unprovisioned-gcc-bazel-experimental-release
linux-arm-jammy-unprovisioned-gcc-bazel-experimental-packaging
linux-arm-jammy-unprovisioned-gcc-wheel-experimental-release
```

I imagine that will also involve creating a new AMI as an unprovisioned base.

One clarification -- Gurobi is not on my critical path.  If it's easy enough to adjust to install the correct architecture, then great!  If not, then it's OK by me to defer that fixup until later.  I can still do a lot of testing & prep work on my end even without Gurobi.

Note that my list of 4 jobs above adds a non-everything job compared to the list of 3 nightlies upthread.  I don't plan on running that one nightly, but it's probably still useful to have around for PR testing.

> The closest graviton equivalent looks to be c6gd.4xlarge.

Confirmed.  I checked https://instances.vantage.sh and that's the most cost-effective instance type for our case, so let's use it for these.

BetsyMcPhail, 2024-03-13:
The jobs specified above and a `Linux ARM` view have been added to Jenkins.

The CI images have NOT been setup yet so these jobs will NOT run!

BetsyMcPhail, 2024-03-13:
The AMI has been created, currently the `init_script` is failing on `update-java-alternatives --jre-headless -s java-1.17.0-openjdk-amd64`  (https://github.com/RobotLocomotion/drake-ci/blob/47a77beae2c6d1a479f75b7f659ba453457b9630/setup/ubuntu/init_script#L115).

jwnimmer-tri, 2024-03-13:
Great progress!  I can see the error you're referring to.

Big picture, the goal here is to get it to the point where Drake users can contribute to making CI run and Drake pass CI.

That basically means we need something like https://drake-jenkins.csail.mit.edu/view/Linux%20ARM/job/linux-arm-jammy-unprovisioned-gcc-bazel-experimental-release/ to have error messages that users can act on by submitting PRs.  As I understand it, error in the init script are only in the private admin console log for that node, not in the typical CI output that users can see.

Assuming that's true, please try to fix the init_script error, so the job runs.  Once we have error logs that users can see, then we'll hand off the remaining work here to the users.

(I think probably you're already on this track, but I just wanted to clarify.)

BetsyMcPhail, 2024-03-14:
We are on the same page. I am hoping that the drake-ci changes will be minimal, I haven't done a search for `amd64` yet but there may be a few other places that need to be updated. I will let you know when I think it's ready to be passed back off to Drake users to work through any build errors.

BetsyMcPhail, 2024-03-18:
@jwnimmer-tri have you given any thought to installing bazel? The CI job (https://drake-jenkins.csail.mit.edu/view/Linux%20ARM/job/linux-arm-jammy-unprovisioned-gcc-bazel-experimental-release/10/console) is now working through the provisioning scripts and is failing because Bazel isn't found (https://github.com/RobotLocomotion/drake-ci/blob/8c8aed8ca1abf3f9fb8b043dcbe6b09e950a6652/driver/platform.cmake#L125).

The install_prereqs does not try to install Bazel on Ubuntun arm64, see https://github.com/RobotLocomotion/drake/blob/38728a99103e0c057e512a701a4df43960d12e19/setup/ubuntu/source_distribution/install_bazel.sh#L56-L65

jwnimmer-tri, 2024-03-18:
That's a problem we're going to hand to our arm64 users and let them solve it.

It looks like the `-release` build is getting past the CI-specific parts now, so this ticket is finished wrt that build.

Can you launch the other three and make sure they make it far enough to start calling Drake setup code?  If they do, then we can close this out as finished.

BetsyMcPhail, 2024-03-19:
All of the arm jobs now launch and make it to the same point of not being able to find Bazel.

closed 2024-03-19 by BetsyMcPhail