RobotLocomotion/drake #21032: [ci] drake-external-examples-linux-jammy-unprovisioned runner

jwnimmer-tri, 2024-02-23:
One thing we all forgot in #20954: in our Jenkins, we need a runner with the label `drake-external-examples-linux-jammy-unprovisioned` for use by https://github.com/RobotLocomotion/drake-external-examples.

BetsyMcPhail, 2024-02-26:
As I was starting to look into this issue, I wondered why drake-external-examples is not using the standard unprovisioned image. While discussing on [slack](https://drakedevelopers.slack.com/archives/C270MN28G/p1708628992063929), the question was raised if the normal jammy unprovisioned build has extra installed that we would not want for external examples.

Looking at the CI documentation, to create a standard unprovisioned image, the following process is followed:

> Unprovisioned images are basically just a stock image. There are apt-get commands in the [init script](https://github.com/RobotLocomotion/drake-ci/blob/main/setup/ubuntu/init_script) that need to be run, but do not run the init script directly.  Paste the commands for
> 
> - `sudo systemctl --quiet disable apt-daily-upgrade.timer apt-daily.timer`
> - Beneath the `-d /media/ephemeral0/ubuntu check`, run the apt-get commands to update and install git, ssh, and jre (needed for jenkins), including the update-java-alternatives command.
> 
> In other words, we are running all of the init_script **except** the creation of `/media/ephemeral0/ubuntu`.

Looking at the `drake-external-examples-linux-focal-unprovisioned` image configuration on Jenkins, it calls this same init_script on startup, so running pieces of that when setting up the AMI will not introduce anything new. 

I compared the configuration for `drake-external-examples-linux-focal-unprovisioned` and `linux-focal-unprovisioned`, as far as I can tell the only difference is the instance cap (8 on external examples vs 64 on standard unprovisioned).

jwnimmer-tri, 2024-02-26:
SGTM

jwnimmer-tri, 2024-02-26:
Yup!  We don't need some new bespoke image.  The regular unprovisioned image seems fine, as best we can tell.

closed 2024-02-26 by jwnimmer-tri

BetsyMcPhail, 2024-02-27:
For completeness, the unused AMI (`drake-external-examples-linux-jammy-unprovisioned`) and Jenkins configuration have been removed.