RobotLocomotion/drake #21041: can't compile with --config=apple_debug

RussTedrake, 2024-02-25:
### What happened?

```
% bazel build --config=apple_debug //solvers/...
```
fails with 
```
ERROR: /private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/external/nlopt_internal/BUILD.bazel:197:11: Compiling src/algs/stogo/stogo.cc failed: undeclared inclusion(s) in rule '@nlopt_internal//:stogo':
this rule is missing dependency declarations for the following files included by 'src/algs/stogo/stogo.cc':
  'external/nlopt_internal/src/algs/stogo/global.h'
  'external/nlopt_internal/src/algs/stogo/tools.h'
  'external/nlopt_internal/src/algs/stogo/linalg.h'
...
ERROR: /private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/external/nlopt_internal/BUILD.bazel:246:11: Compiling src/algs/ags/ags.cc failed: undeclared inclusion(s) in rule '@nlopt_internal//:ags':
this rule is missing dependency declarations for the following files included by 'src/algs/ags/ags.cc':
  'external/nlopt_internal/src/algs/ags/solver.hpp'
  'external/nlopt_internal/src/algs/ags/data_types.hpp'
  'external/nlopt_internal/src/algs/ags/evolvent.hpp'
  'external/nlopt_internal/src/algs/ags/local_optimizer.hpp'
...
```

```
bazel build --config=apple_debug //multibody/...
```
throws those errors and also a bunch from conex
```
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:6:11: error: unknown type name 'TriangularMatrixWorkspace'; did you mean '::conex::TriangularMatrixWorkspace'?
using T = TriangularMatrixWorkspace;
          ^~~~~~~~~~~~~~~~~~~~~~~~~
          ::conex::TriangularMatrixWorkspace
external/conex_internal/conex/triangular_matrix_workspace.h:14:8: note: '::conex::TriangularMatrixWorkspace' declared here
struct TriangularMatrixWorkspace {
       ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:38:1: error: use of undeclared identifier 'TriangularMatrixWorkspace'; did you mean '::conex::TriangularMatrixWorkspace'?
TriangularMatrixWorkspace::TriangularMatrixWorkspace(
^~~~~~~~~~~~~~~~~~~~~~~~~
::conex::TriangularMatrixWorkspace
external/conex_internal/conex/triangular_matrix_workspace.h:14:8: note: '::conex::TriangularMatrixWorkspace' declared here
struct TriangularMatrixWorkspace {
       ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:38:28: error: cannot define or redeclare 'TriangularMatrixWorkspace' here because namespace 'conex' does not enclose namespace 'TriangularMatrixWorkspace'
TriangularMatrixWorkspace::TriangularMatrixWorkspace(
~~~~~~~~~~~~~~~~~~~~~~~~~~~^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:40:22: error: use of undeclared identifier 'supernode_size_'
    : supernode_size(supernode_size_) {
                     ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:41:3: error: use of undeclared identifier 'N'
  N = std::accumulate(supernode_size.begin(), supernode_size.end(), 0);
  ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:41:23: error: use of undeclared identifier 'supernode_size'
  N = std::accumulate(supernode_size.begin(), supernode_size.end(), 0);
                      ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:41:47: error: use of undeclared identifier 'supernode_size'
  N = std::accumulate(supernode_size.begin(), supernode_size.end(), 0);
                                              ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:42:3: error: use of undeclared identifier 'variable_to_supernode_'
  variable_to_supernode_.resize(N);
  ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:42:33: error: use of undeclared identifier 'N'
  variable_to_supernode_.resize(N);
                                ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:43:3: error: use of undeclared identifier 'variable_to_supernode_position_'
  variable_to_supernode_position_.resize(N);
  ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:43:42: error: use of undeclared identifier 'N'
  variable_to_supernode_position_.resize(N);
                                         ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:55:3: error: use of undeclared identifier 'snodes'
  snodes.resize(path_.size());
  ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:55:17: error: use of undeclared identifier 'path_'
  snodes.resize(path_.size());
                ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:56:19: error: use of undeclared identifier 'snodes'
  for (auto& si : snodes) {
                  ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:57:15: error: use of undeclared identifier 'supernode_size'
    si.resize(supernode_size.at(cnt));
              ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:58:25: error: use of undeclared identifier 'supernode_size'
    for (int i = 0; i < supernode_size.at(cnt); i++) {
                        ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:59:18: error: use of undeclared identifier 'N'
      if (var >= N) {
                 ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:62:18: error: use of undeclared identifier 'path_'
      si.at(i) = path_.at(cnt).at(i);
                 ^
bazel-out/darwin_arm64-dbg/bin/external/conex_internal/conex/drake_vendor/conex/triangular_matrix_workspace.cc:63:7: error: use of undeclared identifier 'variable_to_supernode_'
      variable_to_supernode_[var] = cnt;
      ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
```
cc @frankpermenter 

### Version

master

### What operating system are you using?

macOS 14 (Sonoma)

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
russt@TRI-GQ16T2RQ2R drake % brew config
HOMEBREW_VERSION: 4.2.9
ORIGIN: https://github.com/Homebrew/brew
HEAD: e5fefd73cd97cd36ae3af29551f529ae59b333d6
Last commit: 7 days ago
Core tap JSON: 25 Feb 21:15 UTC
Core cask tap JSON: 25 Feb 21:15 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_EDITOR: emacs
HOMEBREW_MAKE_JOBS: 12
Homebrew Ruby: 3.1.4 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.1.4/bin/ruby
CPU: dodeca-core 64-bit arm_blizzard_avalanche
Clang: 15.0.0 build 1500
Git: 2.39.3 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 8.4.0 => /usr/bin/curl
macOS: 14.3.1-arm64
CLT: 15.1.0.0.1.1700200546
Xcode: 15.2
Rosetta 2: false
```

jwnimmer-tri, 2024-02-26:
Most likely, these two lines are the proximate cause of the error message:

https://github.com/RobotLocomotion/drake/blob/efa53019f71c18c5a714200612fac475a0910f7b/tools/cc_toolchain/bazel.rc#L19-L20

Those two settings are (in modern times) particularly problematic.  I guess they were less problematic 5 years ago.  I think for them to work correctly, you need to do the "apple debug" in a completely fresh checkout, not re-use an existing build.  (Or equivalently, `bazel clean --expunge` first.)

Long term, my goal would be to not have `--config=apple_debug` at all.  The question would be if we just do `--compilation_mode=dbg` (aka `-c dbg`) on its own (which is the same as we do on Ubuntu), does lldb debugging still work?

Unfortunately all of the TRI macs are out of service at the moment, so I can't investigate more.

There are some google hits on debugging with dSYM and bazel.  If the plain `-c dbg` doesn't get us what we need, maybe there are some internet tips that would get it working.

jwnimmer-tri, 2024-09-24:
Per #21936, macOS Debug is no longer covered by CI.  It's possible that we should likewise disclaim support for any special Bazel setup to try to make it easier, since we can't automatically test it.

PR #22111 (2024-11-04): [tools] Remove unsupported apple_debug configuration
Closes #21041.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22111)
closed 2024-11-04 by rpoyner-tri