RobotLocomotion/drake #21046: Proper documentation of PartitionConvexSet overload leads to build error in pydrake

sadraddini, 2024-02-26:
### What happened?

`PartitionConvexSet` is not a class method but has two signatures, one with a single `ConvexSet` and one with a vector of it - `ConvexSets`.

This leads pydrake to not build as it apparently does not know which doc to take. Right now the documentation for the second `ConvexSet` has one leading `*` so it actually [does not appear in the documentation](https://drake.mit.edu/doxygen_cxx/namespacedrake_1_1geometry_1_1optimization.html#aeab8ca89c6e95c43cf9ec8c6284f395f), but it should! 

I suggest changing this function's name to `PartitionConvexSets` or find a solution to properly document it. 

+@cohnt 

### Version

79c934b5a8c986c0b5c5481a6b1b0b17632adb69

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
bindings/pydrake/geometry/geometry_py_optimization.cc: In function 'void drake::pydrake::DefineGeometryOptimization(pybind11::module)':
bindings/pydrake/geometry/geometry_py_optimization.cc:1165:57: error: 'const struct<unnamed struct>::<unnamed struct>::<unnamed struct>::<unnamed struct>::<unnamed>' has no member named 'doc'
 1165 |       py::arg("epsilon") = 1e-5, doc.PartitionConvexSet.doc);
      |                                                         ^~~
bindings/pydrake/geometry/geometry_py_optimization.cc:1180:57: error: 'const struct<unnamed struct>::<unnamed struct>::<unnamed struct>::<unnamed struct>::<unnamed>' has no member named 'doc'
 1180 |       py::arg("epsilon") = 1e-5, doc.PartitionConvexSet.doc);
      |                                                         ^~~
Target //bindings/pydrake/geometry:geometry failed to build
Use --verbose_failures to see the command lines of failed build steps.
```
```

jwnimmer-tri, 2024-02-26:
> I suggest changing this function's name to PartitionConvexSets

Don't do this because of docstrings.  If you want to use different names based on the merits, that's fine.  But fixing the docstring stuff is less than 5 minutes of work.

> ... or find a solution to properly document it.

See https://drake.mit.edu/doxygen_cxx/group__python__bindings.html at the heading "Documentation".

sadraddini, 2024-02-26:
> This bug doesn't seem to be present on master -- looks like it's been added somehow in https://github.com/RobotLocomotion/drake/pull/21011. I'll go fix it as part of that pull request.
On master it compiles, but you don't get any documentation for the overload. Check it!

closed 2024-02-28 by sadraddini