RobotLocomotion/drake #21048: Recommendation for setting hydroelastic modulus

DamrongGuoy, 2024-02-26:
Please add to the Hydroelastic User Guide or tutorials on how to set the value of hydroelastic modulus for common materials.

For reference, this table shows Young's moduli of common materials. Is it possible to relate hydroelastic modulus to Young's modulus in a simple layman's way?
https://physicsnet.co.uk/a-level-physics-as-a2/materials/young-modulus/
[image]

The table above shows that rubber's Young's modulus is 0.02 GPa = 2e7 Pa.  Anecdotally, we found that hydroelastic_modulus = 2e7 Pa is too stiff for household rubber objects (size in centimeters).

We have considered in the past the following choices

- hydroelastic_modulus = Young's modulus / 100
  - It gives rubber 2e5 Pa, plastic 2e7 Pa, bone 1.8e8 Pa, copper 1.2e9 Pa, mild steel 2.1e10 Pa.
  - It lacks physics principles.
 
- hydroelastic_modulus = min (Young's modulus, 1e7)
  - It gives everything (rubber, plastic, bone, copper, mild steel) 1e7 Pa.
  - It cannot work with anything stiffer than rubber.
  
As a side note, I was informed that our code might become unstable if we take a hydroelastic modulus greater than 1e7 or 1e8 Pa. If that is true, please document it.

DamrongGuoy, 2024-02-28:
@amcastro-tri gave a very useful analysis of the relation between hydroelastic modulus and Young's modulus. I quote Alejandro here:

_"Hydro and Young's modulus are obviously related, but there relationship is geometry dependent. For instance, we can solve for the elastic force between a sphere and a half-space analytically. This is what you get:
Fhydro = π/3⋅Eₕ⋅d²(3 - 2d/R)
FHertz = 4/3⋅E⋅R½⋅d³⁄₂
where R is the sphere's readius, d is the penetration and I denoted with Eₕ the hydro modulus. As you can see yes, both forces are linear with their respective moduli, but they do not follow the same law. If I change gometries, both laws will change. So you could say that for the same force and penetration, for the sphere/half-space contact, the moduli relate by (for small d/R for simplicty):
 Eₕ = 4/3/π⋅E⋅(R/d)½
As you can see, in general a function of geometry and state. Accoriding to this, a E is a bad approximation at small penetrations and it gets better at larger penetrations.
Maybe we could include an analysis like this in the hydro guide. I could give it a shot if you guys find this analysis useful"_

jbinagia, 2024-03-07:
I like the relationship derived from relating hydroelastic contact to Hertzian theory (at the very least for building up intuition). The factor $R/d$ sort of brings to mind the function [set_penetration_allowance()](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#a1e5e058b968b927f6342a8127aea663a) used to estimate point contact stiffnesses for a desired amount of penetration. Maybe an analogous function could be  created for hydroelastic contact, that takes as input a desired amount of penetration and the Young's modulus for the material and returns an estimate of the corresponding hydroelastic modulus to use?