RobotLocomotion/drake #21059: Clarify doxygen of Meshcat::PublishRecording() about sending objects separately

DamrongGuoy, 2024-02-27:
### What happened?

Currently the documentation says _"... the objects that they modify must be sent to the visualizer separately (e.g. by calling Publish())."_ like this:

https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_meshcat.html#a455ab0b35fdab06f2cc5e0cb9b1b2845
[image]

I'm not sure what _"(e.g. by calling Publish())"_ means.  Is it about the publish event in the system framework? Is it about `SetObject()` ?

When I click on `Publish()`, it went to `drake::lcm::Publish()`, which I think doesn't make sense, like this:

https://drake.mit.edu/doxygen_cxx/namespacedrake_1_1lcm.html#a7f10e39fd5048940095d76f21bdf760d
[image]

jwnimmer-tri, 2024-02-27:
I think that's a leftover from https://github.com/RobotLocomotion/drake/pull/18433.  The animation functions used to be part of the `MeshcatVisualizer` class, which was a LeafSystem with a Publish function.

DamrongGuoy, 2024-02-27:
Thank you, Jeremy.  It's very helpful to see #18433.

CC: @SeanCurtis-TRI .  Thanks for the prior discussion.

closed 2026-11-16 by SeanCurtis-TRI