RobotLocomotion/drake #21091: Compute convex hull of glTF file

SeanCurtis-TRI, 2024-03-04:
`PlanarSceneGraphVisualizer` visualizes complex meshes by computing their convex hull and projecting them onto a plane. As we upgrade our visualization geometries from .obj to .glTF the implementation that computes the convex hull is insufficient. `PlanarSceneGraphVisualizer` currently requires .obj files. glTF is not supported at all.

However, a utility function that can take a `Mesh("foo.gltf", scale)` and create a convex hull representation, bound in python, would be sufficient for `PlanarSceneGraphVisualizer` to be able to make use of glTF files.

Recent PRs (#21012 and #21066) created such a function (albeit in `internal::` namespace). That function operates on .obj and (volume) .vtk files. We just need to extend it to include .gltf files.

jwnimmer-tri, 2024-03-04:
> ... (albeit in internal:: namespace) ...

Perhaps not even worth a mention (and maybe you already know anyway), but FYI: it's valid to bind internal-namespace helper functions into pydrake.  We just do it with a leading underscore.  Then other `pydrake` code can call them, but users are not supposed to.  See e.g. `pydrake.multibody.meshcat._PointContactVisualizer` for prior art.

So, this ticket doesn't imply that we need to promote the API to public.  We can decide that on its own merits.

SeanCurtis-TRI, 2024-03-07:
Definitely worth a mention. I hadn't been aware of that practice. Thanks.

PR #21131 (2024-03-13): MakeConvexHull now supports glTF files
We're using VTK to parse the glTF file. For VTK's parser to work, it needs VTK's OpenGl2 rendering artifacts initialized. This required moving VTK module initialization out of `internal_render_engine_vtk.cc` and into its own compilation unit to be shared.

The convex hull API is exercised in `planar_scenegraph_visualizer`.

While C++'s `MakeConvexHull()` could probably be promoted to part of the stable API, that decision is being deferred for now. It is bound as `_MakeConvexHull`.

Resolves #21091.

- - -
This change is [[image]](https://reviewable.io/rev [... cut]
closed 2024-03-13 by jwnimmer-tri