RobotLocomotion/drake #21095: `model_visualizer` does not support mimic joints or implicit PD controlled joints

dmcconachie-tri, 2024-03-05:
### What happened?

`model_visualizer` does not support modelling choices that are currently only supported by discrete time plants. In one case it throws a warning about ignoring mimic joints, in the other it terminates. This can be reproduced from within anzu via
```
./run --build '' //models:show models/visuomotor/add_prudential_finray_mimic.dmd.yaml
``

### Version

71b39137e0946f2521171866082df6eaffe71577

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
WARNING: /home/dmcconachie/coding/anzu/bazel-bin/models/show.runfiles/anzu/models/food_prep/panda_hand_finray_actuated_mesh_mimic.urdf:134: warning: Joint 'panda_finger_joint2' specifies a mimic element that will be ignored. Mimic elements are currently only supported by MultibodyPlant with a discrete time step and using DiscreteContactSolver::kSap.

...

RuntimeError: Continuous model with PD controlled joint actuators. This feature is only supported for discrete models. Refer to MultibodyPlant's documentation for further details.
```

jwnimmer-tri, 2024-03-05:
FYI #18917 is about having the joint sliders actually account for the mimic (so that the joints are dependently postured).

I guess we can keep this issue separate, just about the throwing part.

jwnimmer-tri, 2024-03-05:
@calderpg-tri Do you remember why `RobotDiagramBuilder` defaults to a continuous-time plant?  That seems like a trap now, since MbP is extremely finicky now about features that are unsupported in continuous-time plants.  Maybe we can change the default to be 1ms or something?

jwnimmer-tri, 2024-03-05:
The `MultibodyPlantConfig::time_step` default is 1ms, so I'd posit that's a good default for the planner as well.

dmcconachie-tri, 2024-03-05:
Thanks; I was looking for the joint actuator part of this and didn't see #18917 during my search. I'm fine with leaving this issue as just tracking the joint actuator part.

calderpg-tri, 2024-03-05:
IIRC, several years ago the predecessor to `RobotDiagram` similarly defaulted to a non-zero timestep due to limitations in the continuous-time capabilities. The ideal answer would be a way to specify that the MbP is not being used for simulation at all, but in the absence of that the discrete-time default sounds reasonable.

jwnimmer-tri, 2024-03-06:
Well, I was hoping this would be a 10-minute fix so I wrote #21098 to get it out of the way quickly.  However, as that CI failure indicates, switching to a discrete time plant for visualization causes the visualizer to throw a _different_ exception now, because it tries to compute inertias even though we're not simulating, and fails some inertia sanity checks.  I guess the continuous time plant was ignoring the invalid inertias, or somehow avoided the error.  I'll put this onto the backlog again, to schedule into a sprint when we have more time available.

MengMengMengH, 2024-04-02:
> The `MultibodyPlantConfig::time_step` default is 1ms, so I'd posit that's a good default for the planner as well.

The problem I have met is that when I used the tag “drake:mimic”，everything goes ok at the beginning.
But when I want to add a lower controller <InverseDynamicsControllers> in my simulation,I was told that I have to use a discrete system,or the drake:mimic joint would be ignored. 
I did set it as a discrete system.But the controller <InverseDynamicsControllers> is a continues system. I have no methods to solve it. Did you meet this problem before? I'm depressed due to this problem.

jwnimmer-tri, 2024-04-05:
Hmm, I didn't actually to end-to-end tests with some of those sample models.  Possibly mimic joint visualization still doesn't work.

jwnimmer-tri, 2024-04-12:
> Possibly mimic joint visualization still doesn't work.

Actually, it's mostly fine.  It doesn't crash anymore, it just prints a warning: `panda_hand_finray_actuated_mesh_mimic.urdf:136: warning: Joint 'panda_finger_joint2' specifies a mimic element that will be ignored. Mimic elements are currently only supported by MultibodyPlant with a discrete time step and using DiscreteContactSolver::kSap`.  Ideally it would actually enforce the mimic and have one less joint slider dof, but that's out of scope here.

PR #21098 (2024-04-04): [planning] Match RobotDiagram default time step to MbP's default
This fixes some crashes when using models that need discrete-time-only modeling features, such as implicit PD control.

Closes #21095.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21098)
closed 2024-04-12 by jwnimmer-tri