RobotLocomotion/drake #21119: Linux AMI refresh

jwnimmer-tri, 2024-03-10:
As of #21103, we have a new version of Bazel installed by the Ubuntu setup scripts.

Please re-bake the Provisioned AMI with that new dependency.

jwnimmer-tri, 2024-03-11:
Scratch that.  We should new AMIs per the original plan -- using Bazel 7.0.2.

There what look to be non-trivial errors with 7.1 shown in the PR logs.

svenevs, 2024-03-12:
drake-linux-jammy-2024-03-12 is built and set to run in Jenkins.  However, since there are currently linux jammy runners active, it will not start getting used until all builds are quiet (a quirk about the jenkins EC2 plugin).

Will post a brief note in #buildcop but if we get at least one round of continuous that uses these (or PRs) then we should be able to close this.

closed 2024-03-12 by jwnimmer-tri