RobotLocomotion/drake #21121: [ci] Try --remote_download_minimal again

jwnimmer-tri, 2024-03-11:
We tried https://github.com/RobotLocomotion/drake-ci/pull/209 but our Bazel 6.x CI wasn't ready for that yet.

Once #21119 images are deployed, we should try re-applying that patch and see if sticks this time.

svenevs, 2024-03-14:
Was looking back into this, the breadcrumbs:

- First attempt: https://github.com/RobotLocomotion/drake-ci/pull/209
- Reverted: https://github.com/RobotLocomotion/drake-ci/pull/210
    - Slack discussion: https://drakedevelopers.slack.com/archives/C270MN28G/p1679539645270689

Recent activity:

- https://github.com/RobotLocomotion/drake-ci/pull/266
- https://github.com/RobotLocomotion/drake-ci/pull/267

My understanding is we want to change

```diff
- --remote_download_outputs=all
+ --remote_download_outputs=minimal
```

But [`--remote_download_output` also has `toplevel`](https://bazel.build/reference/command-line-reference#flag--remote_download_outputs), which may possibly be a better choice.

Regardless of `minimal` or `toplevel`, I'm not actually sure I understand what the failure criteria is (the job logs linked in slack are gone).  Do we just YOLO it and if builds start breaking revert?

svenevs, 2024-03-14:
> Do we just YOLO it and if builds start breaking revert?

Clarification: in https://github.com/RobotLocomotion/drake-ci/pull/209 you can get rough stats, and assuming the test jobs launch for a couple of distros, that would make the `drake-ci` PR theoretically ready to merge.  However, I don't have a record of what failed after we merged it (and therefore had to revert).

jwnimmer-tri, 2024-03-14:
The default in Bazel 7.x was `toplevel`, which is what defeated us.
I think if toplevel didn't work, then "minimal" will also not work.

For now, we should probably put this on hold again until after Bazel 7.2 (which should be soonish).

From https://github.com/bazelbuild/bazel/issues/20161, we might need to use both `--experimental_remote_cache_eviction_retries` and `--experimental_remote_cache_lease_extension`, or else figure out why replaying the build action isn't working.

jwnimmer-tri, 2024-10-22:
After #22066 lands, we should let CI bake for about a ~week to make sure things are stable with 7.4.0, and after that we should try this again.

By "this", I mean landing https://github.com/RobotLocomotion/drake-ci/pull/209 again and watching for fallout.  Remember that the fallout is typically several days later after merging, so we should land it on maybe a Monday and watch for PR flakes the rest of that week.

jwnimmer-tri, 2024-11-11:
We had to revert this again.

jwnimmer-tri, 2025-02-24:
We should try again now with Bazel 8.1.1.  (That means upgrading to use it, first.)

tyler-yankee, 2025-04-25:
We upgraded to use Bazel 8.1.1 in https://github.com/RobotLocomotion/drake/pull/22690, so let's try again.

jwnimmer-tri, 2025-04-25:
I'd say upgrade to 8.2.1 first (which we should do anyway on master).

closed 2025-04-28 by BetsyMcPhail

tyler-yankee, 2025-04-29:
As we've tried this through multiple (major) versions of Bazel now, perhaps better to close as not planned?

BetsyMcPhail, 2026-10-13:
Jenkins logs currently show the following warning:

`[12:11:31 PM]  WARNING: --experimental_remote_cache_lease_extension has no effect since --rewind_lost_inputs is enabled, which recovers from lost remote cache entries as they are encountered.`