RobotLocomotion/drake #21130: Allow `InverseDynamics` with custom plant parameters

nepfaff, 2024-03-12:
**Is your feature request related to a problem? Please describe.**
I want to quickly test inverse dynamics control tracking with different plant parameters (e.g., mass). I can change these parameters in the plant context. However, `InverseDynamics` creates a default context of the controller plant (see [here](https://github.com/RobotLocomotion/drake/blob/master/systems/controllers/inverse_dynamics.cc#L39)).

**Describe the solution you'd like**
I want some way to change the parameters used by `InverseDynamics` programmatically. My suggested solution is for `InverseDynamics` to take `plant_context` as an additional optional argument.

**Describe alternatives you've considered**
An alternative solution would be to add methods for changing all the default values of interest. Some of these already exist. See [here](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_joint.html#a5683d2bda6da395fd555c77061d9f816) for an example. However, most of them, such as `RigidBody` parameters, don't yet exist.

**Additional context**
I'd happily open a PR for adding the `plant_context` argument. Are there any issues with this approach?

sherm1, 2024-03-14:
Assigned to @jwnimmer-tri for disposition.

jwnimmer-tri, 2024-03-14:
=> +@hongkai-dai -@jwnimmer-tri as component lead

nepfaff, 2024-03-19:
I gave the `plant_context` as an optional argument approach a shot [here](https://github.com/nepfaff/drake/tree/add_plant_context_argument_to_inverse_dynamics). However, it currently fails with a segmentation fault and I'm not yet sure why.

hongkai-dai, 2024-03-21:
> I want some way to change the parameters used by InverseDynamics programmatically. My suggested solution is for InverseDynamics to take plant_context as an additional optional argument and propagating this through to InverseDynamicsController.

I think that makes sense. Would you mind creating a C++ test so that we can debug the segmentation fault?

PR #21287 (2024-04-23): Add `plant_context` argument to `InverseDynamics` and `InverseDynamicsController`
Closes #21130

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21287)
closed 2024-04-23 by jwnimmer-tri