RobotLocomotion/drake #21133: [framework] Unneeded state update notifications propagated to systems with no state

sherm1, 2024-03-13:
This is a performance issue where unneeded invalidations lead to unneeded re-computations.

Background:
- The Simulator operates on the whole Diagram without knowing its substructure
- That includes integration if there is _any_ continuous system
- After integration of the _diagram_, all subsystems are notified that their continuous variables (q,v,z) may have changed
- That notification is sent regardless of whether a particular subsystem actually has any continuous variables
- A discrete MultibodyPlant does not have any continuous variables

Problem:
- Some MbP cache entries (notably hydroelastic contact surfaces) declare dependency on "configuration"
- Configuration is defined unconditionally to include both discrete and continuous variables
- Hence the integration step incorrectly invalidates MbP's "configuration"
- So hydro contact surfaces get recalculated unnecessarily next time they are needed

This problem is most noticeable with MultibodyPlant but we are sending notifications of state changes to all subsystems, regardless of whether they actually have state. That's a waste of time also.

jwnimmer-tri, 2024-03-13:
For extra safeguarding, I wonder if we should change MultibodyTreeSystem when in discrete mode to unsubscribe its configuration ticket from the irrelevant trackers entirely.  (Just until #9171 is fixed, after which I can declare things more naturally.)

It's fine for SystemBase to be conservative with the default dependency graph, but MbTS knows exactly what it's configuration state is, so it can be precise with its dependency graph.  I worry that there are more traps than just this one particular control flow path hit by the Simulator.

I still like the #21134 to mitigate the Simulator-Integrator invalidation problem for any discrete-state-only LeafSystem.  I think maybe MbP is important enough to be perfect that editing the graph directly might be worthwhile.

sherm1, 2024-03-14:
> I wonder if we should change MultibodyTreeSystem when in discrete mode to unsubscribe ... 

For now I think it is enough to rely on the general framework improvement in #21134 rather than making bespoke changes to MbP. We can revisit if there is a new problem.

PR #21134 (2024-03-14): Disable unused dependency trackers for better performance
This PR prevents "out of date" notifications for state variable changes from being propagated in systems that don't have the requisite state variables.

There are just two changes:
- We add a "suppress_notifications" flag to DependencyTracker that prevents further propagation of notifications to subscribers
- At Context allocation, we suppress notifications from dependency trackers for non-existent state variables.

This prevents, for example, a "continuous state variables changed" notification from affecting configuration_ticket in a s [... cut]
closed 2024-03-14 by sherm1